Ultimaker/Cura #3998: Support generation isn't beginning until level 2

IRONCUTTER, 2018-06-29:
**Application Version**

3.4.0
**Platform**

Windows 10 home 64-bit
**Printer**

Custom - Anet A8- Reprap
**Steps to Reproduce**

Any print where support is requested, it does not begin until z value of .4
**Actual Results**

Support doesn't print on 1st layer so it doesn't stick
**Expected results**

**Additional Information**

smartavionics, 2018-06-29:
Hi, please provide a project file that shows this problem - without that info it's impossible to get an understanding of what is going wrong.

diegopradogesto, 2018-06-29:
Yes, please send a Project File (File -> Save Project...) so we can figure out what happen.

IRONCUTTER, 2018-06-29:
Attached is the file as you requested.  Is there a CFG file of Ultimaker's settings that you would like?
Bill Porter

smartavionics, 2018-06-29:
> Attached is the file as you requested. Is there a CFG file of Ultimaker's settings that you would like?

Sorry, mail attachments don't get through to github, please attach the file as it says below. Thanks.

IRONCUTTER, 2018-06-29:
Good morning, Mark.  I have attached an STL and G-code file that demonstrate.  Support structure dies not begin until level 2 on my machine.  If this is due to a setting, please advise.

Thanks,
Bill P

smartavionics, 2018-06-29:
Bill, please attach the files to the web page. Attaching them to emails does not get them onto github or send them to me. Thanks.

IRONCUTTER, 2018-06-29:
Mark,
Here are the files

[Printer midbrace.zip](https://github.com/Ultimaker/Cura/files/2149099/Printer.midbrace.zip)

Also, when I flip the model, support dies generate.  It appears to only be a problem when the support must generate "in captivity", under the model in the enclosed cavity.

IRONCUTTER, 2018-06-29:
Done.  Please advise if you need further info or settings.

smartavionics, 2018-06-29:
Hi Bill, what I really need is the project file, not the gcode or STL files (thanks anyway for the those as I can see that the support isn't starting on layer 1). So please save the project, rename (or zip) the 3mf file and attach to this issue page. Thanks.

IRONCUTTER, 2018-06-29:
[Printer midbrace.zip](https://github.com/Ultimaker/Cura/files/2150742/Printer.midbrace.zip)
Mark, here is the source file.  Hope you can deal with an OBJ file, it was done on Tinkercad and this was the only option other than STL.

smartavionics, 2018-06-29:
Sorry, I can't unzip that file. But, look, I don't want the OBJ, I want the .curaproject.3mf file that you get when you save the project from Cura. In there will be the model and all the settings so I can try and work out why the support is not starting on the bed. If you can't supply the project file, I can't help you.

IRONCUTTER, 2018-06-29:
[CFDMP_Printer midbrace.zip](https://github.com/Ultimaker/Cura/files/2150954/CFDMP_Printer.midbrace.zip)
Got you.  Here you go.

smartavionics, 2018-06-29:
No, that's not a project file, it doesn't contain any settings. Please do File->Save Project and create a .curaproject.3mf file.

IRONCUTTER, 2018-06-29:
[CFDMP_Printer midbrace.curaproject.zip](https://github.com/Ultimaker/Cura/files/2151177/CFDMP_Printer.midbrace.curaproject.zip)
This seems to be everything

smartavionics, 2018-06-30:
Hi, thanks for that project file. I now understand what's happening. It's a bug that is being caused by the fact that you have brim enabled and also "brim only on outside". The support is inside the object and there is no brim underneath it but the support generator must think there is brim below.

There are several workarounds. You can either turn off the brim only on the outside option. Or you could use a skirt instead of a brim. If you do that you can make the skirt very close to the object and so it acts just like a brim as I show below. Hope this helps.

[image]

IRONCUTTER, 2018-06-30:
> <div id="AOLMsgPart_2_f76044cd-4dd1-4e7d-9769-fdcb3d17bd9d">
> <div class="aolReplacedBody"><p>Hi, thanks for that project file. I now understand what's happening. It's a bug that is being caused by the fact that you have brim enabled and also "brim only on outside". The support is inside the object and there is no brim underneath it but the support generator must think there is brim below.</p>
> <p>There are several workarounds. You can either turn off the brim only on the outside option. Or you could use a skirt instead of a brim. If you do that you can make the skirt very close to the object and so it acts just like a brim as I show below. Hope this helps.</p>
> <p>[image]</p>
>
> <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—
>
> </div>
> </div>

Sounds good, Mark.  Thanks.
Bill P

Sent from AOL Mobile Mail

smartavionics, 2018-07-01:
Hi @diegopradogesto, can we please reopen this issue as there is a bug that needs fixing. The problem, as mentioned above is that the support is being removed from the first layer when brim is enabled and the brim is only on the outside (not inside the hole where the support is). Thanks.

diegopradogesto, 2018-07-02:
I've just discussed with the rest of the team and we are adding this bug to our board. @IRONCUTTER now you can use the workaround until we fix it.

diegopradogesto, 2018-07-02:
Devs. CURA-5525

ianpaschal, 2018-07-10:
@diegopradogesto you mentioned in #4046 that this is marked as fixed in Jira. Shall we mark this ticket as fixed and close it in that case?

diegopradogesto, 2018-07-10:
Yes, is marked as fixed, so I close the issue. It would be fixed of course in the next release.

closed 2018-07-10 by diegopradogesto