Ultimaker/Cura #4000: Cura is Cool!

fieldOfView, 2018-06-29:
I just wanted to mention that. 
Feel free to close this issue.

**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

fieldOfView, 2018-06-29:
I could not let issue #4000 be just some random issue.
🍰 🍾 🍪

Appesteijn, 2018-06-29:
w00t w00t! Thanks :)

diegopradogesto, 2018-06-29:
I agree!!!

Ghostkeeper, 2018-06-29:
Duplicate of https://github.com/Ultimaker/Cura/issues/3796?

filipgoc, 2018-07-04:
Hahaha. This issue should be tagged as a _constant Improvement_

closed 2018-11-26 by fieldOfView

nallath, 2018-11-27:
Cura is no longer cool :(

fieldOfView, 2018-11-27:
> 1. To make clear we (the Cura team) is not going to do this and don't give any false hopes that we will do this in some distant future. Of course someone else can implement this, but we don't have the time.
> 2. To clean up the issue list so we can focus on the things that are critical and need to be fixed.