Ultimaker/Cura #4004: Create option for Gyroid infill

probonopd, 2018-06-30:
Please create gyroid infill that is the shape as graphene. This shape has been shown to be very strong as a 3D printed object and if used as an infill would presumably create very strong 3D prints

http://itsybitsymag.com/2017/01/11/mit-research-ten-times-strong-lightweight-graphene/

Reference:
https://github.com/prusa3d/Slic3r/issues/92

ianpaschal, 2018-07-02:
FYI the link doesn't work.

Kriechi, 2018-07-02:
It was working a few days ago...
Anyway: https://www.reddit.com/r/3Dprinting/comments/81b39y/new_infill_pattern_gyroid_looks_amazing/

filipgoc, 2018-07-04:
Nice.

Theagainmen, 2018-07-17:
Any news on this? Will Cura add this? :)

ianpaschal, 2018-07-18:
For now I don't think so, but we will continue to discuss as a team. We would probably be open to adding it via a pull request if someone writes the engine code for it.

Theagainmen, 2018-07-18:
Thanks, keep it up, love Cura!

probonopd, 2018-07-27:
Here is a semi-scientific on-camera comparison:
https://www.youtube.com/watch?v=upELI0HmzHc

tmbdev, 2018-07-31:
Gyroid infill is very useful not just because of its mechanical properties, but also because the interior of the part is fully connected; this is true for no other infill pattern AFAIK. Fully connected interior infill space allows the part to be filled easily with epoxy, and on the other hand lets liquid/air escape through a single hole.

fieldOfView, 2018-07-31:
Thanks for providing a usecase other than "I want it".

tmbdev, 2018-08-02:
Let me summarize also the "traiditional" justifications for gyroid: gyroid is one of the strongest infills for a given weight, it is fairly isotropic (=works about the same in all directions), doesn't require strong accelerations, and prints fairly fast. I don't have much hands-on experience with it yet (since Cura is my primary slicer), but I think it may make a good default just based on its mechanical properties alone.

Snypes88, 2018-08-08:
Missing this feature is absolutely the only reason I use slicer PE!! It is a must if you would like to print PETG infill at any decent speeds as... Come on dev team, this is a chance to make Cura the greatest slicer again 👍

smartavionics, 2018-08-08:
> It is a must if you would like to print PETG infill at any decent speeds as...

As what?

Snypes88, 2018-08-16:
@smartavionics you left out the remainder of the sentence. Sharp changes in direction for line infill result in more blobbing and thus poor print quality. Besides, everyone loves Gyroid, so it should just be implemented 😄

smartavionics, 2018-08-16:
> @smartavionics you left out the remainder of the sentence. 

LOL, you added the remainder of the sentence after I made my comment!

ianpaschal, 2018-09-05:
Cura has some big stuff in the works so unfortunately for now, no. This issue is marked as "Deferred" because we do not plan to devote any time to it in the foreseeable future, but would like it implemented eventually. Pull requests are very welcome.

smartavionics, 2018-09-11:
OK, I'll have a go at implementing it.

smartavionics, 2018-09-11:
Could some kind person attach a simple model and also the gcode that has been created from that model using gyroid infill so I can compare it with I will be producing. Cheers.

agibson2, 2018-09-11:
I will when I get home tonight.  A simple cube with walls and top bottom layers?  How big do you want it?

Snypes88, 2018-09-11:
Here's a simple 25x25x25mm cube using gyroid infill 👍 

[Cube.zip](https://github.com/Ultimaker/Cura/files/2373010/Cube.zip)

Snypes88, 2018-09-11:
Here's the same cube but only with infill... in case its easier for you to check that way @smartavionics 

[CubeInfillOnly.zip](https://github.com/Ultimaker/Cura/files/2373047/CubeInfillOnly.zip)

agibson2, 2018-09-12:
You might want to mention what infill percentage you used.

Snypes88, 2018-09-12:
an STL of a 25mm cube..... 🤔😉 The infill is at 17% for reference

agibson2, 2018-09-12:
This is a 25x25 cube also with a bunch of different infills with and without walls/top/bottom.  I also scaled it by 1200% and did a 300x300 cube with a 2% infill.  I couldn't include it as it is bigger than 10MB.  I included that in the pic because I printed a 300x300 2% infill for fun... well I aborted it after about 36mm.  The 2% 300x300 actually printed which I was surprised by.

The link for the 300x300 is below if you are curious.
https://drive.google.com/file/d/1OB2GbjhJTXPmezXLHlKC8ajgqa7sbjWD/view

[image]
[cube_25x25.zip](https://github.com/Ultimaker/Cura/files/2373244/cube_25x25.zip)

smartavionics, 2018-09-12:
Thanks for the files, they helped a lot.

I am happy to report that after bashing my head against sufficient walls, progress has been made.

As the following pics show, the basic functionality is now working. Needs to be tidied up and performance could be a problem. With high density infill, the slice time goes up hugely. Does Slic3r also slow down a lot for dense gyroid infill?

[image]

[image]

[image]

fieldOfView, 2018-09-12:
Awesome!

We could implement a warning for the infill density setting in the frontend if the infill pattern is set to gyroid.

smartavionics, 2018-09-12:
Hi Aldo. That could be a good idea but it's early days yet so I want to find out where the time is going and get it as fast as possible first.

Do people actually use this infill very dense? I just printed that little cube and it seems very sturdy.

Later tonight I will upload a new Linux dev archive so that if anyone is using Linux they could try out the current state of the art.

smartavionics, 2018-09-12:
How strong would you like your infill, Sir?

Gyroid with infill multiplier set to 3...

[image]

smartavionics, 2018-09-12:
I have uploaded a new Linux archive that includes this feature in its current early state. If anyone tries it out, please provide feedback.

The archive can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

agibson2, 2018-09-12:
I can't test the Linux build unfortunately but as for speed, Yea it does get slower on another slicer.  This with testing with cubes.  I assume more complex stuff will take longer.

Testing on the other slicer software...
100x100x100 mm cube using different infill
20% 1 to 2 seconds
40% 3 seconds
60% 4 seconds
70% 5 seconds
80% 6 seconds
90% 8 seconds
95% 9 seconds
99% 9.5 seconds
100% Gui popup... The Gyroid infill pattern is not supposed to work at 100% density  Shall I switch to rectilinear fill pattern?

agibson2, 2018-09-13:
I got my CentOS 7 system up and running again.  I tried to run the Linux archive you provided on CentOS 7 but I get an error about  libssl.

File "/home/user1/Downloads/cura/lib/python3.5/ssl.py", line 98, in <module>
  import _ssl                        # if we can't import it, let the error propagate
ImportError: libssl.so.1.0.0: cannot open shared object file: No such file or directory

CentOS 7 has libssl.so.1.0.2k   (openssl-1.0.2k-12.el7.x86_64)

Closest thing I could find which is a recent thing...
https://github.com/Ultimaker/Cura/issues/4061

If I shouldn't post that info in this thread as it is unrelated to gyroid, let me know and I will remove it.

smartavionics, 2018-09-13:
Sorry, the archive didn't work for you, it was created on a Ubuntu 16.04.5 system.

With regard to the infill density could you please create a 50% infill cube using another slicer and attach the gcode so I can compare densities.

Cura does have a performance problem when the infill has many lines so some work is needed there.

Another issue at the moment is that gyroid infill doesn't work with the connect infill lines option and it would be nice to get that working.

agibson2, 2018-09-13:
My first print.  It is pretty strong.  The thickness of the lines are thinner but that is likely some setting differences between slicers.  I am completely new at the other slicer.

Some things that are noticeable...  In Cura it stops extruding when it moves over to the next line.  With the other slicer, it just continues to extrude when it moves to the next line which gives it a little more rigidity and looks cleaner (no stringing at top speeds as it never stops extruding when doing infill) but wastes more filament of course.  I actually like Cura's way because I want to print some things without top bottom and walls and the extra filament messes up the pretty structure on the sides :).  It would be better to have an option though so we could decide if we should just keep extruding or stop during the movement to the next line.  With the Cura version, the cube *looks* kind of not so perfect though because of the stringing as that is most visible when you look at it but that might be my settings and it is only visual.  If you look at the inside though, It looks nice.  I have never cared about infill looks until this version of infill so that just might be me :).

I am liking it.  It is a very good start for this.  Thanks for putting the time into it.  I am going to keep using this version for Gyroid infill even though I have to go through hoops to get the files to my desktop to print them :).

[image-3 2](https://user-images.githubusercontent.com/6733850/45521721-4161cb80-b78d-11e8-96b5-4115a4d85727.jpg)
[image-2 9](https://user-images.githubusercontent.com/6733850/45521722-4161cb80-b78d-11e8-9700-8d996b1c27aa.jpg)
[image-1 26](https://user-images.githubusercontent.com/6733850/45521723-41fa6200-b78d-11e8-8247-d96069c23eb1.jpg)

agibson2, 2018-09-13:
I forgot to mention that the 7.6% in the other slicer is about the same as 10% in Cura going by the pattern size.  Not that any one is right or wrong, just an observation.

agibson2, 2018-09-14:
The other slicer seems to be overextruding which is why it is thicker and stronger.  Cura is right at 0.4mm (my nozzle width) but slic3r is making the lines from 0.52 to 0.56.  I think I have slic3r configured properly but I can't be 100% sure.  Maybe they are overextruding on purpose with a default setting somewhere or maybe just this infill or maybe something else entirely :).

agibson2, 2018-09-14:
Increasing the Infill Line Width to 0.5mm (0.4mm nozzle) in Cura makes a 10% infill match the 10% infill pattern of slic3r.  That seems to indicate they are overextruding.  I have never used the infill line width before so that is new to me.

agibson2, 2018-09-14:
300x300x36.2 2% infill 0.54 Infill Line Width (0.4mm nozzle).  Assuming 0.54 does anything with a 0.4mm nozzle.  The print turned out nice except it started to have some lifting issues as the line ends are not connected as mentioned above.  I used some blue tape on one small section to keep it going and it recovered.  It looks nice and pretty strong considering it is only 2% infill.  I doubt 2% infill will be the norm though as it doesn't have enough coverage for a good top layer of course.  It looks nice though :).
[img_1385](https://user-images.githubusercontent.com/6733850/45547771-f0cf8a00-b7ef-11e8-9560-ee61ac28449e.jpg)

probonopd, 2018-09-14:
Thanks for your testing and comparison. This is very helpful @agibson2

smartavionics, 2018-09-14:
Yes, I was just about to comment, thanks for the testing.

I have been busy again today and apart from a slew of bug fixes and performance improvements, have also implemented the connect lines feature for the gyroid infill as you can see here:

[image]

I found it quite challenging to implement but it appears to work OK on the few examples I have tried. Further testing and feedback would be most welcome.

smartavionics, 2018-09-14:
In about 20mins or so, a new archive should appear on my Dropbox which contains the day's release.

fieldOfView, 2018-09-14:
https://www.youtube.com/watch?v=E-WHW-QNswE

Theagainmen, 2018-09-14:
@smartavionics legend man! Can't wait to see this. (Lines in layer view look pretty trippy but that's just the way it is) thanks! :)

smartavionics, 2018-09-14:
That's what we want, trippy prints!

I think the infill line connections are not yet perfect as sometimes it connects lines that don't have to be connected but, overall, it's working pretty well as you can see...

[image]

smartavionics, 2018-09-15:
Hi Adam, thanks again for the super testing and feedback. Those pics are looking good.

With regard to the density, what I have done is adjust the pitch of the gyroid lines based on the infill line distance scaled by some factor. The infill line distance is related to density and line width. For example, if you are using simple "lines" pattern then 50% density would use a line spacing equal to the line width. 25% density would use a line spacing 3 times the line width. The gyroid pitch is calculated so that the total length of filament required for the infill is close to the length required for the lines pattern with the same infill density. The idea here is that changing the infill pattern from lines to gyroid should result in parts with similar weight.

I think that the layer start point is fixed so it won't start the next layer from where the current layer finishes. I can see the benefit of doing less travel but the downside would be that it would immediately reverse and extrude on top of the still molten layer below so it could give quality issues?

agibson2, 2018-09-15:
Thanks for taking the time for the explanation.  It is interesting to hear the reasons for things.  It sounds very logical.

For the start point, I looked at the gcode from slic3r and it does start right back over top of where it finished the last layer.  Maybe that isn't always the case... I certainly don't know.  I just thought it was interesting as it was faster and didn't have as much stringing on my printer but stringing can be compensated for mostly.  Maybe a Cura dev already investigated that at some point and determined it is bad.  I am sure you Cura devs are doing what is best for Cura.

I printed another object "Save pangolins" but with outer layers you can't see anything inside so no since in showing a pic of it.  It turned out to be an excellent print.
https://www.thingiverse.com/thing:2064359/

I think people are going to be very happy with what you have done when this eventually gets approved.  It is working well for me as it is now.  I think it is going to be a popular feature when it eventually gets released.

agibson2, 2018-09-16:
You have been very busy lately according to the pull request commits today.  Will another build be up today?  If so I will wait for my next print.

smartavionics, 2018-09-16:
OK, I've fixed the issue but the build won't be uploaded for a couple of hours as our internet is tied up for a while.

Snypes88, 2018-09-16:
Amazing progress 😍 Never thought things would go so fast! @smartavionics When can we expect to try out your progress under Windows? Or is the Alpha channel only for Linux?

smartavionics, 2018-09-16:
Sorry, @Snypes88 , I can only create Linux versions of my Cura as I don't use Windows here. All of the source is available on github so someone who can build Windows releases could build it but I'm afraid I don't know anyone who does build for Windows from source.

I don't know if it would work but if you have sufficient free disk and memory you could install a small Linux system using Virtualbox and try running my releases using that.

agibson2, 2018-09-17:
Another print using Gyroid.  I like the pattern.  It is a consistent pattern from the side compared to the triangles that I always used before.  This is a pencil holder.

[image-2 12](https://user-images.githubusercontent.com/6733850/45603335-3c08b900-b9f9-11e8-8126-a7e851b8d41d.jpg)
[image-1 30](https://user-images.githubusercontent.com/6733850/45603336-3ca14f80-b9f9-11e8-897f-27ceac4ba201.jpg)

smartavionics, 2018-09-17:
Hi Adam, thanks for all the feedback, I'm glad you're happy with the results.

> One thing I noticed when increasing Infill Line Width (from a previous print) from 0.4 mm to 0.6 mm for a 0.4 mm nozzlie is that it shows as 2 lines when really it is just a thicker single line. I tried triangle infill and it showed the same thing so I guess that is expected?

Hmm, I don't see that can you provide some screenshots of the layer view?

agibson2, 2018-09-17:
I think it is just a rendering/view thing with Cura on Ubuntu inside a HyperV (Windows 10).  Regular Cura on Windows doesn't exhibit this weird looking rendering.

[screenshot](https://user-images.githubusercontent.com/6733850/45640309-4752f780-ba80-11e8-9a82-3e1adc926a73.png)

smartavionics, 2018-09-17:
Ah, that screenshot is showing the layer view to be using compatibility mode. The OpenGL in the VM must have a low version. I don't think there really are two lines there.

agibson2, 2018-09-17:
I agree.  It is just using whatever defaults HyperV and Ubuntu use.  I am not worried about it.

Here is another print using transparent PLA.  Really liking the pattern.
[img_1413](https://user-images.githubusercontent.com/6733850/45653905-92343580-baa7-11e8-86e4-4ec6a0401857.jpg)
[img_1412](https://user-images.githubusercontent.com/6733850/45653906-92343580-baa7-11e8-9e28-db16ec2f3771.jpg)

agibson2, 2018-09-24:
> Cura has some big stuff in the works so unfortunately for now, no. This issue is marked as "Deferred" because we do not plan to devote any time to it in the foreseeable future, but would like it implemented eventually. Pull requests are very welcome.

@ianpaschal Any chance of this getting out of deferred status with the work and the pull request that has been done on it by smartavionics?

probonopd, 2018-09-24:
He says

> Pull requests are very welcome.

👍

fieldOfView, 2018-09-24:
Note that the "deferred" status is only meant to show that UM devs won't spent time developing this. Fortunately for us, @smartavionics is not an UM dev.

ianpaschal, 2018-09-24:
Yup! "Deferred" means we want it (or in the case of a bug, it _should_ be fixed) but that we don't have the time/dev power to make it happen for the foreseeable future.

When Mr. Avionics has made a PR for us to review, we'll slap an "In Progress" label on the PR. 😉

smartavionics, 2018-09-24:
> When Mr. Avionics has made a PR for us to review, we'll slap an "In Progress" label on the PR. wink

https://github.com/Ultimaker/Cura/pull/4365

https://github.com/Ultimaker/CuraEngine/pull/850

Over to you...!

ianpaschal, 2018-09-24:
Aha oops! I'm actually now officially working only the UM3 Networking Plugin, so I can't speak to whether or not anyone in Cura team has picked it up but I imagine @Appesteijn will put it in a sprint soon in the near future.

smartavionics, 2018-09-24:
> Isn't Open Source great?

Sure is. It makes a lot stuff possible that otherwise would be very difficult to achieve.

[17 more replies]

closed 2018-10-02 by ChrisTerBeke