Ultimaker/Cura #4005: Cura 3.4 Crash with complex objects in layer_height printer definition

thetrebor, 2018-06-30:
**Application Version**

3.4.0

**Platform**

OSX

**Printer**

Monoprice Mini Delta

**Steps to Reproduce**

In any printer definition file xxx_def.json have a complex object as the layer_height.
For example: in the key overrides.layer_height let ```"layer_height": { "minimum_value": 0.05, "maximum_value": "machine_nozzle_size * 0.75", "maximum_value_warning": "machine_nozzle_size * 0.5 + 0.01",  "default_value": 0.10 , "value": 0.10 },```
Crash with error
```Traceback (most recent call last):
File "/Users/ultimaker/build/3.4/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 898, in getQualityProfilesDropDownMenuModel
File "/Users/ultimaker/build/3.4/build/inst/lib/python3.5/site-packages/cura/Machines/Models/QualityProfilesDropDownMenuModel.py", line 47, in init
File "/Users/ultimaker/build/3.4/build/inst/lib/python3.5/site-packages/cura/Machines/Models/QualityProfilesDropDownMenuModel.py", line 70, in _update
File "/Users/ultimaker/build/3.4/build/inst/lib/python3.5/site-packages/cura/Machines/Models/QualityProfilesDropDownMenuModel.py", line 107, in _fetchLayerHeight
TypeError: float() argument must be a string or a number, not 'SettingFunction'```

If you let ```"layer_height": {"default_value":0.10}``` then everything works.
This setting worked in previous versions.

**Actual Results**
<!-- What happens after the above steps have been followed -->
Application crash with complex object in layer_height settings

**Expected results**
<!-- What should happen after the above steps have been followed -->
It should work as in 3.3

**Additional Information**
<!-- Extra information relevant to the issue, like screenshots -->
Have a great day!

LipuFei, 2018-07-02:
Hi @thetrebor , could you try changing your `value` field to have double quotes like this?
```
"value": "0.10"
```

fieldOfView, 2018-07-02:
To be clear: "value" should always be a string (that will be interpreted by Python), and "default_value" should be the literal value (a float in this case).

thetrebor, 2018-07-02:
I changed the setting to as follows
```"layer_height": { "minimum_value": 0.05, "maximum_value": "machine_nozzle_size * 0.75", "maximum_value_warning": "machine_nozzle_size * 0.5 + 0.01", "default_value": 0.10 , "value": "0.10" },```
Crash with same message.
Then I changed the setting to:
```"layer_height": { "minimum_value": "0.05", "maximum_value": "machine_nozzle_size * 0.75", "maximum_value_warning": "machine_nozzle_size * 0.5 + 0.01", "default_value": "0.10" , "value": "0.10" },```
Crash with same message.

(note, that I did try these things before submitting the bug report but I did them again per your instructions)

I just tried to eliminate and re-add the values one at a time.  The presence of the "value" key crashes it irregardless of the type of the expression being stored.

crashes with:
``` "layer_height": { "value": "" },
"layer_height": { "value": 0.10 },
"layer_height": { "value": "0.10" },
"layer_height": { "value": "NULL" },```

unusual behavior, causes "configuration error, your profile is corrupt":
``` "layer_height": { "value": NULL },```

To be clear this is new behavior and this profile worked OK in the previous version.  Everyone who's complaining about Cura not working with the MonoPrice mini delta probably is using the same profile from the MPMD wiki with this setting.

thetrebor, 2018-07-23:
I see this is still flagged as "need info".
Is there any more info you need?

ianpaschal, 2018-08-01:
I don't see any printer definition with key `value`, everyone I checked (20 or so at random) all use exclusively `default_value`.

Probably this should be handled more graciously than a CTD but in the mean time, can you try it without the `value` key?

thetrebor, 2018-08-01:
it works when you remove the "value" key.
it used to work with the "value" key prior to 3.4
an example of a profile with the key "in the wild" in userspace is here: https://mpminidelta.com/slicers/cura
This is a popular site with a large number of users.  We can of course, change the definitions, but a cardinal rule of development is "don't break userspace"... especially without warning to users with a CTD and as a unintended consequence from a different change.

I couldn't find the data dictionary for the printer json.  Is it documented somewhere what the valid keys are for the printer definitions?  Looking through the commits I see that there are at least some printer/keys that use "value" as the setting.  For example:

resources/definitions/ultimaker2.def.json, line 28
"machine_start_gcode" : { "value": "\"\"

Users would be expected, sans documentation, to assume that this is a reasonable setting.

ianpaschal, 2018-08-01:
I'm looking through the 3.3 branch and that also uses exclusively `default_value` with the `layer_height` property.

Same goes for 3.2.

Why `value` worked, I'm not sure. Maybe because it's referencing another key, not a value. It's also being used for `layer_height_0`, not `layer_height` and I wouldn't just assume because one setting has a valid key that that key must be valid other places too.

I'll admit our documentation is somewhere between not good and non-existent in this area and we've actually been working on that lately (specifically for quality files). But if one looks at existing profiles, including the one you linked to, they *all* use `default_value` to set the `layer_height`. I don't think this is a case of changing the API willy-nilly.

thetrebor, 2018-08-01:
I do appreciate you looking into it.  If you take the profile from the site i referenced and install it you'll notice cura CTD (and it didn't used to in <3.4).  Remove the "value" key from the json and you'll notice it no longer crashes in >=3.4  
I don't know what other data to provide to assert it's an (unfortunate) outcome of a recent api change.  I can spread the word to people that we should change the printer definition on the wiki and in the entire installed user base, but in the squishy meat brains of users it will look like Cura is the problem and not the profile.

ianpaschal, 2018-08-01:
Yeah of course the code has changed, but my point is it's why, for example, JS UI frameworks say *"Never ever call your UI components single-word names because you can't predict if it will be added to the next HTML spec!!!!11"*.

Unfortunately we don't have a clear spec but I don't consider something that wasn't our standard practice but happened to work (also did it actually work or was it just successfully ignored without CTD?) as being something that needed to be preserved. There's no way to know all the various ways in which people on the Internet might have constructed a profile in the past and make sure we never make a change which breaks these unknown profiles.

And yes, we need a spec so that there's a "right" way to do it, but the mean time this can be avoided by submitting a PR. Then we know of a profile's existence, we can either be aware that some defs are constructed in such and such way, or offer feedback not to do it in that way.

ianpaschal, 2018-08-01:
I know, and we appreciate your effort. This is a really good bug report, and it would be great if more people put that kind of effort in (some don't even manage to answer the 5 or so questions in the template).

That being said, we're a pretty small team and while we will make sure to review every PR that comes in, we don't have the manpower to go to companies who are our competitors and poke them to fix things.

I'll be sure to talk to the team about handling unknown keys more gracefully. The other actions we are taking are working on improving our documentation and working on setting up a better infrastructure for both users and companies to create these kind of files and offer them as downloads (although can't say too much about that just yet I think).

But again, thank you for taking the time to do an investigation into the matter.

closed 2018-08-01 by ianpaschal

thetrebor, 2018-08-01:
Good luck.

Ghostkeeper, 2018-09-03:
I fixed this crash for the next release here: https://github.com/Ultimaker/Cura/commit/6df029971d2f0dc4070959c892f2d79c3cf1cb04

greg-ware, 2018-11-07:
I have same issue on QidiTech profiles. Thanks to @thetrebor , @ianpaschal  , @Ghostkeeper for investigating and fixing this!

commit 6df02997 (2018-09-03): Don't crash if layer height is a formula

We obtain the layer height here but that might be a formula so we should evaluate that formula here.

Fixes #4005.