Ultimaker/Cura #4006: Cura 3.4 Slicing freeze

OverSavior, 2018-06-30:
Cura 3.4 No plugins

Windows 10 Pro x64

Custom FDM printer

1. Load the model
2. Change the scale
3. Press "Prepare"

Cura start slicing the model, fill slicing bar a bit and then slicing bar become clear and nothing happens. Even if you wait for hours. It happens sometimes with all the models I'm trying to prepare. Same models works ok on another pc (on that pc such issue never happen)

Model had to be sliced

https://youtu.be/BqXAvyoDtNQ

OverSavior, 2018-06-30:
I tried to reinstall the Cura. Also I disabled firewall and antivirus. Still Cura not slicing.

ianpaschal, 2018-07-02:
> Same models works ok on another pc (on that pc such issue never happen)

With the same version in Cura? Then it seems there's definitely something wrong with that computer. We won't be able to recreate or work on the bug if it only occurs on one of your machines. 😕

Can you share your log files? Perhaps there is an answer in there.

OverSavior, 2018-07-02:
Well, yesterday solved the problem : )
Seems that Avast did something to Cura files, dunno. I tried to reinstall Cura again, but this time I also deleted Cura folders from AppData. Now it works perfectly!

closed 2018-07-04 by alekseisasin

GreatBizarro, 2018-09-16:
I have the same problem in 3.4, it says slicing but nothing happens past that. Windows  7 pro, Intel I7 3.07ghz 24gb mem. , I have put Zonealarm on a 5min timeout and AVG stopped File Shield and Behavior Shield. No progress bar and just sits till I hit cancel
[cura.log](https://github.com/Ultimaker/Cura/files/2386826/cura.log)

ianpaschal, 2018-09-19:
Your log shows that Cura can't connect to Cura Engine. Likely it's being blocked by these services. Please try unblocking Cura or at least turning them off temporarily to confirm that that is the issue.

GreatBizarro, 2018-09-22:
Thanks, I tried setting zonealarm to 5min time out in the app but that didnt do it just removed it and replaced it with another firewall. Works now.