Ultimaker/Cura #4008: slicing problem

HenkV74, 2018-07-01:
**Application Version**
3.4.0

**Platform**
windows 10

**Printer**
anet a8

**Steps to Reproduce**
load the file and turn the round knob so that the widest side is on the plate

**Actual Results**
The brims around the other rods disappear and the knob is the only one with a brim

**Expected results**
Everything should have a brim

**Additional Information**

[curaerror.zip](https://github.com/Ultimaker/Cura/files/2152756/curaerror.zip)

[image]

smartavionics, 2018-07-01:
If I raise the (rotated) round item enough, it then slices OK (I think). Looks like a bug...

[image]

HenkV74, 2018-07-01:
The only way I can get the brim to appear is when "generate support" is enabled. Moving the objects, partal overlap or Z resetting does not change a thing. For now i've printed via 2 files, no problem but this looks like an easy to debug file. Thats why I shared. Looks definitely like a bug to me.

smartavionics, 2018-07-01:
Hold on, I think I know what's happening here. The problem is that the long things have very sharp edges on the bottom. They are so sharp that they are thinner than the minimum printed wall thickness and so no walls are printed and the objects no longer have anything on the lowermost layers. Look at the first 3 layers when all objects have z set to 0...

[image]
[image]
[image]

So it's not a bug in Cura, the problem is that the model has detail too fine to be printed (obviously depending on layer height, line widths, etc.)

HenkV74, 2018-07-01:
The problem is that the position of the round knob triggers this behaviour. Without the knob it generates the brims without problem, with the knob it generates only if the knob is upright. With the knob flat it does not generate brims. The rods have not been touched in these 3 cases. Not the behaviour I think is logic from an user point of view. So I consider it a bug, I only saw it when my objects did not stay on the plate during printing because there where no brims.

smartavionics, 2018-07-01:
What Cura does is lower the entire model so that the first layer is not empty. When the knob is on its side, it also has a very thin edge and so that gets thrown away in the same way that the thin edges of the long items do so (by good fortune,depending on layer height, line width, etc.) all of the items have some content on the first layer and so the brim/skirt will go around them all. But when the knob is orientated flat side down, it is the only item that will produce some printable lines on the first layer and Cura will not sink it into the build plate so that the other items are also producing printable lines on the first layer.

Whether this behaviour should be considered a bug or not is arguable. Personally, I would not really class it as a bug, more of an interaction between features (removal of unprintable lines and removal of empty lower layers). Also, it could be argued that the problem is really with the model inasmuch that the model has more detail than is printable using the current print parameters (layer height, line width, etc.)

Anyway, I'm just trying to be helpful here, it's not for me to say whether this is a bug that should be fixed or not. Cheers!

ianpaschal, 2018-07-04:
@smartavionics is right about the behavior, so this isn't a bug, and is working as expected.

As for what to do about it; the answer is to chamfer or fillet your model. You should never be trying to manufacture anything with razor edges like that. If this was any thing else: casting, cnc-milling, such edges would break off anyway. You can get around it by turning the piece over but in truth its just bad practice as far as industrial design goes. If you look closely virtually no edges on any product are ever without at least a 0.2 mm fillet or so.

I'm going to consider this question answered.

closed 2018-07-04 by ianpaschal

HenkV74, 2018-07-04:
Ok, I was thinking that Cura would manage that and that the smallest base line would be generated to avoid printing in the air for the first layer. It doesn't but it is not consequent in the way it does this when the item is printed alone or printed with other items. I live with it and take care to avoid this in the future. It still is something that has to be explained to users like me.