Ultimaker/Cura #4009: BUG; brim only on outside; turning it of increased print time by a day

UltiArjan, 2018-07-01:
**Application Version**
3.4

**Platform**

**Printer**
UM3 / UMs5

load a model, default settings, toggle the "Brim Only on Outside"﻿ checkbox.

**Actual Results**
see screenshot, it wil add many hours to a simple print of a few minutes.

strange thing is it results in a new crazy print time every time you toggle the button again.

**Expected results**
add just a few minutes of print time

BUG was reported on the forum by klaus_kreamer here:
https://community.ultimaker.com/topic/23583-cura-34-brim-only-on-outside/

[image]

[image]

Vandrasc, 2018-07-02:
Yes, reproduced on other models as well, it seems that we generate random slicing times each time a slicing is retriggered.
Thank you!

AbeFM, 2018-07-02:
I've seen this a bunch, across a lot of models from a lot of sources.

Generally with multiple build plates turned on, not sure if it's related.

Vandrasc, 2018-07-03:
Devs, see CURA-5533

SimonRafferty, 2018-07-16:
Also having this issue in 3.4.0 & 3.4.1
It's refusing to print an object I printed OK a couple of weeks ago.  Unfortunately, it needs the brim on the inside to be successful.
[image]

PimpTrizkit, 2018-07-24:
I second this issue.

Anycubic i3 Mega, Cura 3.4.1. 

Im trying to print the Articulating Phone Mount: https://www.thingiverse.com/thing:2448971
(i first noticed it on the "Arm - Threaded.stl" file)

PT

diegopradogesto, 2018-07-25:
This is now fixed!

closed 2018-07-25 by diegopradogesto

SimonRafferty, 2018-08-07:
>This is now fixed!
In which version?  It's happening in 3.4.1 which appears to be the latest.
[image]

diegopradogesto, 2018-08-07:
Sorry for the confusion. It's fixed in master branch, that will be the next release. So, you need to wait a bit until we release the new version.