Ultimaker/Cura #4010: Custom Material Profiles not showing up after import sucessfully

braga-dev, 2018-07-02:
**Application Version**
3.4.0

**Platform**
Windows

**Printer**
Custom FDM Printer

**Steps to Reproduce**
When creating  a custom fdm printer I want to add my custom materials profile.

**Actual Results**
When I use the import function, Cura says it successfully imported the material profile but it doesn't shows up in my material list. It only shows the basic profile for each material and Ultimaker Material Profiles.

**Expected results**
Show a Custom tab where i have all my custom material profile.

LipuFei, 2018-07-03:
Hi @braga-dev , The material list only shows the materials with diameters compatible with your current printer's extruder. For example, if your currently active extruder has a material diameter of 2.85, but the imported material has 1.75, it won't show up because it's not compatible. You can go to "Machine Settings" to your extruder and change your nozzle size to see if your material shows up.

braga-dev, 2018-07-03:
Hi @LipuFei i just noticed there is a slight bug or a mis-understanding on my part. So when I create my printer I said the nozzle diameter was "1,75" instead of "1.75" and the difference between using a dot ou a comma was the problem in the appearing profiles. Nevertheless when creating a material profile we use 1,75 with a comma which induced me in error.

LipuFei, 2018-07-03:
Sounds like a bug, we will check that. Thanks for the report!

Ghostkeeper, 2018-07-06:
I can't reproduce this bug. At least, not here on Linux.
* In the Material Manager the diameter is localised and so I can only use commas (as per the Dutch localisation). Typing a period doesn't add a period; nothing happens.
* In the Machine Manager it says `2.85` by default. Typing `1,75` is possible changes immediately into `1.75` when focus is removed from the text box. And then the 1.75mm materials load fine.

Ghostkeeper, 2018-08-17:
Did it reproduce on Windows?

LipuFei, 2018-08-17:
I didn't have a chance to try this on Windows...

Ghostkeeper, 2018-10-18:
@Vandrasc Could you try this on Windows? Using a comma instead of a period for the filament diameter?

closed 2020-06-20 by no-response[bot]

Ghostkeeper, 2020-06-30:
We've since determined that this is caused by the user's locale. If the operating system is in a locale that is using commas as delimiter (e.g. German) then you also need to use commas in the machine settings dialogue.