Ultimaker/Cura #4011: [3.4.0] Arranging models for one at a time printing fails

upatamby, 2018-07-02:
**Application Version**
Ultimaker Cura 3.4.0

**Platform**
Windows 10 Pro

**Printer**
Ultimaker 3 Extended

**Steps to Reproduce**
Laod body.stl in Cura. 
Select Extruder 2 for this model.
Rotate the model 90 degrees anti clockwise using the green circle (this is the X/Z-plane).
Multiply the model 8 times.
In my profile I used PETG (based upon a CPE-profile), but I don't think the material is important here.
Don't use any support and no raft or brim.
Choose Special Modes, Print Sequence = One at a Time.
**Actual Results**
The models are ordered in 3 rows and 3 columns.
Seen from above:
[image]

Start printing.

The print starts with the model in the left column, front row.
1.
x x x
x x x
X x x
2.
x x x
X x x
X x x
3.
X x x
X x x
X x x
So far so good.
But then comes:
4.
X x x
X x x
X X x
While printing the 4. model the printhead touches the already printed model left from it.
It made the whole build plate waggling, so I aborted the printers menu.
And then - this is a severe issue!!! - the printhead moved straight back to its home position (the back corner at the left) without first lowering the build plate, so it jammed to the printed models and got stuck before reaching the corner because a model got in the way and then the printhead started to make a very bad noise. I switched off the printer and hope nothing got damaged.

**Expected results**
First of all: the arrangement of the models should be correct so the models and printhead should never collide. Second (but very important: if the print ends or aborts the buildplate should lower itself before the printhead starts moving.

Vandrasc, 2018-07-02:
@jackha , is this related to the other arrange ticket(Cura-5468) or a different issue? Thank you

jackha, 2018-07-03:
1) Aborting does not work correctly, which is a known issue in the firmware team. Unfortunately it hasn't been picked up yet EM-2284
2) Ah! I think it's a bug of Extruder 2 + one at a time. I'll check.

jackha, 2018-07-03:
One at a time bug confirmed. Currently testing if it only affects Extruder 2, or that the big convex hull doesn't take skirt and brim into account. Or a combination of those.
Devs: CURA-5538

Ghostkeeper, 2018-08-15:
This issue has been fixed for the next release. Thanks for the report.

closed 2018-08-15 by Ghostkeeper