Ultimaker/Cura #4012: Will not save updated print over old file

astecme, 2018-07-02:
Cura 3.4.0 will not overwrite an existing file on an SD card. It says it has but does not do it, the file remains the same.

diegopradogesto, 2018-07-03:
Is it maybe because you don't have write permissions? Are you able to save a file for the first time from Cura? Can you provide the log files?

astecme, 2018-07-03:
Ah no. Sorry I dont have log files however I changed the orientation of the print and swapped from skirt to brim. I did this a few times, the last time I checked the file on the card and the timestamp had updated but the content was the same. Deleted all files on the card and did the save again and it printed fine.

diegopradogesto, 2018-07-04:
Is there something we can help with? Otherwise I will close the issue.

closed 2018-07-06 by ianpaschal