Ultimaker/Cura #4013: Printing via Usb issue ( stop printing mid print )

sakkklim, 2018-07-02:
**Application Version**
<3.0.4

**Platform**
<Macboo pro Touch Bar / Mac OS Mojave beta.2

**Printer**
<Jgaurora a5

**Steps to Reproduce**
<when print via usb using Cura 3.0.4 + mojave beta

**Actual Results**
<stop printing when first layer. 

**Additional Information**
<It always seems at first layer. after 1st layer. It's almost well-working.
### **I think Cura should stabilize the Usb printing with marlin firmware.** 
Another Program (. Repetier , Pronterface ) doesn't stops.
I re-installed  Cura latest and old version. latest version of Cura works more badly.
USB printing is very convenient function to use .  :)

ianpaschal, 2018-07-03:
This has been reported before in various forms.

closed 2018-07-03 by ianpaschal

sakkklim, 2018-07-03:
what's the reason of this situation?

ianpaschal, 2018-07-03:
We're not sure, but we are aware *of* the situation. Problem is that it's very hard to test when we don't have people's actual computes with their actual USB connections and their 3rd party printers.

sakkklim, 2018-07-03:
that's interesting.
after I updated Mac OS beta version (Mojave 2) this situation got worse.
I wish they can figure it out. 
It's really comfortable function . =)