Ultimaker/Cura #4016: Cura crashes when saving Gcode

Richardacosta7, 2018-07-03:
**Application Version**
3.4.0

**Platform**
Windows 10 Home

**Printer**
CR-10

**Steps to Reproduce**
Hit save to file 

**Actual Results**
Crashes every time 

**Expected results**
Save G code to file

**Additional Information**
This is a new HP computer 7/3/2018 had (Windows 10 S). I changed it to (Windows 10 Home) to install Cura. I been having one issue after another one. Cura seems to be working fine only crashes when saving to file. Please help!

diegopradogesto, 2018-07-03:
Does it happens with every model? Can you send us the log files to see what happened?

Richardacosta7, 2018-07-03:
I get it everytime. I tried all types of models. I'll send the log files
when I get home. Thanks, Richard

Richardacosta7, 2018-07-03:
[cura.log](https://github.com/Ultimaker/Cura/files/2160987/cura.log)

diegopradogesto, 2018-07-05:
I had been told that there were similar issues when having shell extensions running in windows at the same time. Maybe @Ghostkeeper has another clue.

alekseisasin, 2018-07-05:
@Richardacosta7 , even probably any other installed application might affect to the crash. can you check **event viewer** in windows, it will show additional information about system crashes.

Richardacosta7, 2018-07-07:
[Log files.zip](https://github.com/Ultimaker/Cura/files/2172388/Log.files.zip)
Still can't get Cura to save. I reinstalled window. Attached is the windows log files,
Thanks,
Richard

wcj97, 2018-07-07:
I'm having the same issue. Cura also crashes when I attempt to make setting changes or change the visibility of slice settings. Some guys in the Ultimaker forums looked at my Cura logs and couldn't figure out why it crashed.

Richardacosta7, 2018-07-11:
Please help. For some reason Cura does not run on windows 10. I returned the brand new computer I got with windows 10 s that I could get Cura to save on. I bought another one with windows 10 home edition and the something happen it crashes on saving GCode. diegopradogesto any luck with the log files? Thanks, Richard

YURI0869, 2018-07-14:
Hello, I am having the same problem with my HP windows 10 but not on my acer windows 10. However, upon doing what diegopradogesto's suggestion, I was able to use the latest version and save Gcode files without crashing. Thanks diegopradogesto.

alekseisasin, 2018-07-16:
Hi @YURI0869 , @Richardacosta7 , may I ask you to do some testings for us. We already have a fix for this issue, but we are not sure whether it works or not, we just need a more testing on different machines. If I will give you a link to a special Cura build and you will try to reproduce the issue. In this case, if the file can be saved without any problem then we will be sure that this issue is solved and the fix will be in the next Cura release version. 

After installing Cura beta version the Cura private data might be corrupted, if this will happen you will need only re-install stable Cura version.

wcj97, 2018-07-18:
@alekseisasin Can I also get a link for the beta build? I’m having the same issue, plus freezing on some settings changes. I’m interested to see if this fixes both my freezing issues.

alekseisasin, 2018-07-18:
Yes, sure. @wcj97 you can download it from here: https://we.tl/JcfqXHz7E5

wcj97, 2018-07-20:
The beta link did not work for me. I am still having the same performance issues.

YURI0869, 2018-07-23:
@alekseisasin. Sorry I just saw your message it was on spam folder. What exactly do you want me to do? I have a monoprice mini select printer.  On my CURA V 3.4.1 I can save GCode files without CURA shutting down.  Anything I can do to help, let me know

YURI0869, 2018-07-23:
@diegopradogesto diegopradogesto, Hi! was it the Python you want me install? If it is, I did download python 2.6.4 and enable the send print info, it didn't work.  I also installed the Cura 3.4.99 master and did enabled send print, it did not work as well.

wcj97, 2018-07-23:
@diegopradogesto @yurio869, I’m curious, did either of you have a python interpreter such as Pycharm or Anaconda installed on your computers?

YURI0869, 2018-07-23:
@diegopradogesto. No,  even when I uninstalled the Python, it did not work. However on my Acer windows 10, Cura works just fine with or without phython and with enabled send print. I think it has something to do with the security settings of HP.

YURI0869, 2018-07-23:
@wcj97. No,  even when I uninstalled the Python, it did not work. However on my Acer windows 10, Cura works just fine with or without phython and with enabled send print. I think it has something to do with the security settings of HP.

Ghostkeeper, 2018-09-03:
This might be a duplicate of #1537 #2633 #2644 #2731 #4220 (not sure yet).

Megdiver, 2018-10-23:
I am running Cura on an ACEPC GK1 with Windows 10 home and when I installed 3.5.1 it would crash when I tried to save g-code.
I went back to 3.4.1 and I can save without crashing.
Is this a reoccurring issue similar than before?
Any suggestions?
I have uninstalled 3.5.1 for the time being but would love to be able to use the newer version.

bigsexyjoe, 2018-11-19:
What solved this issue for me was trying a different microSD card. I think the one I was using may have been corrupted. I hope this helps someone!

Maybe the Cura devs could put a warning/detection in around this.

wcj97, 2018-11-19:
@bigsexyjoe I don’t think it’s the MicroSd card. Or at least, it’s definitely not in my case. I can typically save to the SD card alright, but if I try and save to my computer or another location, the code write fails. It usually just leaves off the “End GCode”, but writes everything else okay. Cura also crashes though, so I’m not sure what it is about writing the End GCode that causes that.

diegopradogesto, 2018-11-19:
@wcj97 could you please attach the entire configuration folder? Also a project file.

YURI0869, 2018-11-20:
I think it is all about your computer settings or firewall maybe. Mine did not work on HP but works fine on Acer with the send print info is checked. But works fine when it is not on HP if the send print info is unchecked.

Ahabs-Wale, 2018-12-03:
We were struggeling also with Cura like the thread opener, also on a brand new HP laptop, Win 10 Pro/64. Everything updated to most recent versions.
We installed, deinstalled, cleaned everything over days, took logs in Windows and Cura, but no hint why Cura (several versions tested) shuts down itself when saving the gcode files. 
The only hint was the missing “End GCode” and everything after that code in the file.

Once we found you and this topic we also unchecked the "sending file" option - and now everything is working fine! We still don't have any clue why this influences the simple file operation of saving, but saving the gcode works now like a charme.

Many thanks to all who were helping here!

closed 2019-01-07 by nallath

LipuFei, 2019-06-20:
Hi @wcj97 , did you disable the anonymous data option in the release you mentioned? Could you try Cura 4.1 with the anonymous data option on and see if that works for you? Thanks!