Ultimaker/Cura #4019: [3.4.0] Where is the source code release?

wilriker, 2018-07-04:
**Application Version**
3.4.0

**Platform**
Arch Linux

---

Actually this is not a bug report I just wonder where the GitHub source code release/tag for 3.4 is. Arch Linux packages around Cura are based on these releases and as long as there is no source code download I assume there won't be an update to the packages. So when can we expect this?

alekseisasin, 2018-07-05:
Do you mean this one : https://github.com/Ultimaker/Cura/tree/3.4

wilriker, 2018-07-05:
No, I mean one that would be visible and downloadable at https://github.com/Ultimaker/Cura/releases (also CuraEngine and the other relevant repos)

ChrisTerBeke, 2018-07-09:
Check the documentation and Wiki in this repository, all the answers are in there.

wilriker, 2018-07-09:
@ChrisTerBeke Can you tell me where exactly the Wiki tells me where to find the source tar-ball for 3.4? It tells me how to install the current master (I know how to do that) from Git. I know I could also change the branch before running `make` (or `cmake`) but this is about a fixed downloadable release that will be used to create a OS-specific package - as they are available up until 3.3.1.

At least I have not found any mention regarding a change. But I am more than happy to be corrected.

ChrisTerBeke, 2018-07-09:
I'm gonna give you 1 link: https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source. Everything else is just learning how GitHub works...

fieldOfView, 2018-07-09:
Chris, are you even reading what @wilriker asks? He clearly knows how git works. There is no tagged release of the 3.4 branch, though there have been releases of every branch before it.

wilriker, 2018-07-10:
Now that this issue has been closed, can I expect an answer anymore?

ianpaschal, 2018-07-10:
@alekseisasin @ChrisTerBeke Guys, we know how GitHub works. Source code releases can be found [here](https://github.com/Ultimaker/Cura/releases) and it's true that we haven't put 3.4 up yet.

ianpaschal, 2018-07-10:
Today we are releasing 3.4.1. Unfortunately we're going to skip uploading the 3.4 release. If you want it. you can get it on the [Ultimaker website](https://ultimaker.com/en/products/ultimaker-cura-software/list).

wilriker, 2018-07-10:
@ianpaschal Thanks! I am only interested in the latest released version, so 3.4.1 will be absolutely fine for me. :+1:

Just seen that Cura and CuraEngine are already on the releases pages with 3.4.1 so I can close this. :-)

closed 2018-07-10 by wilriker