Ultimaker/Cura #4025: Support Blocker being ignored by Tree Supports

teknokraze, 2018-07-04:
Not sure why, but using the tree support on this model, (see attached pic), is creating support where it's not needed. I've added a support blocker, trying both on the bed and on the top where the picture indicates but the tree support is ignoring it.

Any ideas?

Thanks...

**Additional Information**

[image]

ianpaschal, 2018-07-05:
Have you gotten the support blocker to work with tree support in other places? I haven't tested yet for myself but it's very possible that since they use very different generation code, the tree support doesn't respect the support blocker.

ChrisTerBeke, 2018-07-06:
Tree Support is in experimental, so it's expected that things like this happen. This would simply be added to our list of things to do to take it out of experimental.

teknokraze, 2018-07-06:
Just discovered today a possibly neat benefit for this bug. Turning on both support types, both regular and trees, I can use the support blocker to keep regular supports from generating where I want tree supports while maintaining better support coverage where I want it on the low slope parts. Kinda cool!

ianpaschal, 2018-07-16:
Devs, see CURA-5575 on JIRA.

Ghostkeeper, 2018-08-27:
The bug in this case is not (I believe) that the support blocker is being ignored. It's more that it seems to ignore holes in the overhang polygons. I've seen this as well.

Ghostkeeper, 2018-11-20:
Our project manager removed this issue from our planning. We are unable to spend time on it in the near future.

TracKer, 2019-12-28:
Problem is still reproducible. Will there be a solution?

nallath, 2020-09-08:
We forgot to close this. The issue should be resolved in 4.7

Herya, 2020-10-23:
The problem is still there in 4.7.1
[image]

Ghostkeeper, 2020-10-27:
@Herya could you share a project file that reproduces that problem?

I'm currently unable to reproduce the issue. It seems to work fine.
[image]

Herya, 2020-10-27:
Happened to me with this model : https://www.thingiverse.com/thing:3622530

It wants to make tree support inside a cavity, even though I ticked the "touching buildplate" option.
So I wanted to block it with the blockers, but as you saw on my screenshots, it didn't work.

Ghostkeeper, 2020-10-28:
Also with that model, it's properly blocking support as far as I can see:
[image]

Could you provide us with a project file? The project file also contains the profiles and settings you're using and where you placed that support blocker. You can create one using File -> Save Project...

Dunngeon, 2020-11-14:
I believe this issue still persists in 4.7.1, I have a similar issue to the user above. Even with support blocks blocking the tree support, it still extends down into the blocked volume. See attached images.
[image]
[image]

teknokraze, 2020-11-14:
You know, a guy opens a issue like this to one, have it hopefully fixed, and two, hope it helps others having the same problem. I think it's not a good system to automatically "close" an issue that still seems to be there simply because the original poster doesn't contribute more. Incidentally, the issue does seem to be fixed for the model I originally posted. I suggest to others that you open your own support thread because, apparently, they'd be happier with multiple repeating issue threads rather than dealing with them within an already established thread.

Ghostkeeper, 2020-11-17:
We're perfectly happy with other people than the original poster providing the project file, of course. Doesn't matter who posts the issue really. Without a project file I'll always make a reasonable effort to reproduce the bug myself but if the reproduction steps don't reproduce the issue, there's very little I can do. We leave the known problems open, of course. But to prevent confusion as to which issues are actually issues and which issues are not, we have a window of response of 2 weeks. It's just a clean-up. Really it's still a manual action because the bot will only close the issue when the status label is attached. See adding that Status: Needs Info label as the closing action (the template warns you that we'll close the issue if it's not filled in), but just closed with a delay of 2 weeks.

Ghostkeeper, 2020-11-17:
This seems to be working as intended. That project file indeed reproduces the symptom you point out:
[image]
However the support blocker only blocks support where it's being generated. It doesn't prevent support from passing through the model if it starts somewhere up higher.

The original issue from Teknokraze was different though. The issue was that support was being generated to support a hole. And the support blocker had no effect on the hole either. Since we fixed the support to no longer support empty space of a hole, this issue doesn't occur any more either. So we consider this issue fixed then.

closed 2020-11-17 by Ghostkeeper