Ultimaker/Cura #4030: [3.4.0] Automatic slicing wait time should be increased or user customisable

inkychris, 2018-07-05:
**Application Version**
3.4.0

**Platform**
Windows 10

**Current Behaviour**
Changing any custom print setup value often triggers automatic slicing before completing a modification of a setting eg. typing a longer value or hesitating slightly.

**Preferred results**
The automatic slicing should be triggered after a longer period of time, preferably within a user specified amount of time so as to avoid upsetting those who don't feel this is an issue?

alekseisasin, 2018-07-05:
Hi @inkychris , the simple solution is to find "Slice automatically" setting in Cura preferences and disable it.

inkychris, 2018-07-05:
That is the simple solution; however, most of the time it's nice to have it enabled, it's just that it starts trying to slice too soon, ie. while you're still editing a field, which then also makes the UI somewhat laggy.

inkychris, 2018-07-05:
I suppose to be more specific, the automatic slicing should trigger sometime after you stop interacting with the UI rather than sometime after you start.

alekseisasin, 2018-07-06:
I am not sure this solution will satisfy other users. I think if the plugin will be done for another your ticket it will help you.

inkychris, 2018-07-06:
Adjusting the wait time is perhaps not the right approach but the negative impact on the performance of the UI when it starts slicing is at least worth noting. If I adjust infill percentage for example, it has already started slicing before I've finished deleting the contents of the percentage box which then makes typing the value back in laggy, resulting in sometimes putting the wrong value in and then compounding the issue.

The plugin for #4031 will definitely help me out but that is more for longer changes like arranging duplicate objects on the build-plate rather than changing specific settings in the print setup.

JohnEdwa, 2018-08-05:
As a recent example of mine:

Go to change 'Horizontal Expansion" setting which is currently set to "-0.08".
Click the value box, press 'CTRL+A' to select all, press 'backspace' and the box will clear.
Cura will almost immediately deselect the box, fill it with "0",  and start an auto slice.
Select the text box again, press '-'.
You now have ~0.5 seconds to press '0', or Cura will reset it again to '0' and start an auto slice.

Ghostkeeper, 2018-09-28:
> You now have ~0.5 seconds to press '0', or Cura will reset it again to '0' and start an auto slice.

Good guess! We do already have a waiting time before starting an auto-slice after any modification. It is set to 500ms.

inkychris, 2018-09-28:
As it happens I am now experiencing this for 3.4.1. Not sure if it was fixed or a fresh install sorted it.