Ultimaker/Cura #4031: [3.4.0] Automatic slicing toggle box should be added to main interface

inkychris, 2018-07-05:
**Application Version**
3.4.0

**Platform**
Windows 10

**Issue**
Sometimes it can be useful to toggle automatic slicing on or off just briefly, or sometimes repeatedly when moving parts around etc. It would be helpful to have the automatic slicing toggle box present in the main UI, perhaps near the slicing progress bar/print time/save to file area to more quickly toggle between automatic and manual slicing. This would save having to open up the general settings window each time.

fieldOfView, 2018-07-06:
I’ll write a plugin for that. Remind me if I haven’t posted a link here in a weeks time.

fieldOfView, 2018-07-08:
I have repurposed a plugin that provided this functionality in Cura 2.4:
https://github.com/fieldOfView/PauseBackendPlugin

For ease of testing, here's a curaplugin file, that you can drag and drop into a Cura window:
http://files.fieldofview.com/cura/v4/PauseBackendPlugin.curaplugin

inkychris, 2018-07-08:
Cura didn't seem to want to load the curaplugin file but I cloned your repo into my plugins directory and it works perfectly. Thank you very much!

fieldOfView, 2018-07-08:
Ah, I see, the .curaplugin format is no longer accepted.
Good to know you got it to work.

Shall I publish this to the official plugin repository?

inkychris, 2018-07-08:
I haven't found any issues with it so it would be great to have it on the official repo. Has my vote!

fieldOfView, 2018-07-13:
Any suggestion what to use instead of pause & play?

diegopradogesto, 2018-07-13:
I was thinking more on a checkbox or a switch, but not totally sure. I will think during the weekend and comment on your repo if I have a better idea.

fieldOfView, 2018-07-13:
I don't want it to take up much space; there's already so little room there (especially in eg the german locale).

diegopradogesto, 2018-07-16:
@inkychris we decided not to implement this directly on Cura. However the PauseBackendPlugin will work perfectly for your purpose. Hope we can add it soon to our toolbox.

@fieldOfView I was talking with @AnneUltimaker about it and she also thinks that a toggle button is better than the Play/Pause. What about adding it below the "Prepare" button? Is that possible?

inkychris, 2018-07-16:
It makes sense to leave this as a plugin. I would also second it being a toggle box like in the main settings menu, although the benefit of the pause/play is that it doesn't require a label to infer what it does which might take up more space?

I feel like the best place for the toggle box would be above the progress bar, on the right, at the end of where it says "Ready to slice" etc.

fieldOfView, 2018-07-16:
Without making changes to Cura, to the left of the Prepare/Cancel/Save/Print buttons is the only place I can add a control.

Placing a control below the Prepare button would take vertical space, and I am not a fan of taking up even more vertical space than the controls in the sidebar already do.

closed 2018-08-15 by inkychris