Ultimaker/Cura #4033: [3.4.0] Stuck at the heating step

kirilledelman, 2018-07-05:
**Application Version**
3.4.0

**Platform**
Mac

**Printer**
Custom printer with 2 extruders:
[Dual_Extruder_Calibration_Print-Calibration_Target.3mf.zip](https://github.com/Ultimaker/Cura/files/2167418/Dual_Extruder_Calibration_Print-Calibration_Target.3mf.zip)

**Steps to Reproduce**
Load two models, set each to different extruders, hit print via USB.

**Actual Results**
Bed heats up first, then second extruder. Then nothing happens, it just sits there forever with bed at 70, and E2 at 200, and E1 at 0. Printer info says "E2 heating".

**Expected results**
Expecting E1 to heat to 200 (or 130 - its standby temp), and printing to begin.

**Additional Information**
Saving .gcode and printing from Pronterface works fine, no preheating issues.

A-Smith, 2018-07-08:
I have what I think is the same issue with an Ultimaker 3:

**Application**
3.4.0

**Platform**
Windows 10

**Printer**
Ultimaker 3 (firmware version 4.3.3.20180529)

**Steps to Reproduce**
Load a single model. Set it to use the second extruder. Disable the first extruder. Print via network.

**Actual Results**
The bed and second extruder heat up, but the first one does not. The LCD on the printer is stuck at "Heating PrintCores...", with the progress bar about a quarter of the way across the screen. Reading the temperature through the network API (via "api/v1/printer/heads") shows:
- printhead 0
  - current = 33.2
  - target = 175
- printhead 1
  - current = 175.3
  - target = 175

The first core is not heating up. That's ok - I only want to print this model with the second. But it looks like the printer is waiting for it. I get the same result when trying to print via USB drive.

**Expected Results**
The printer should ignore the first printcore and start printing once the second reaches the required temperature.

diegopradogesto, 2018-07-18:
@A-Smith I just tested with a UM3 with the same firmware and it worked. Can you provide the Project file that failed and I can see if it works?

diegopradogesto, 2018-07-19:
OK, thanks @A-Smith.

Regarding the original issue, I'm taking a look at the gcode output generated by CuraEngine and it seems correct to me:

`...`
`;Generated with Cura_SteamEngine 3.4.1`
`T1`    <-- Select the second extruder (T1) since it is the first that prints something
`M190 S70`   <-- Set BP temp
`M104 S130`
`M104 T1 S200`
`M109 S130`
`M109 T1 S200`   <-- Wait until T1 reaches 200 ºC
`M82 ;absolute extrusion mode`
`G28 ;Home`
`...`
`G0 X73.016 Y177.617`
`M104 T0 S200`     <-- This is line 5512. While T1 is printing, T0 is starting to heat up.
`G1 F1800 X73.918 Y177.858 E254.34502`
`...`
`G92 E0`
`T0`    <-- This is line 6098. The T1 finishes and T0 (extruder 1) is selected. 
`G92 E0`
`M109 S200`  <-- Wait until 200 ºC is reached in T0
`M104 T1 S130`   <-- Set T1 to the standby temperature
`G0 F3600 X118.424 Y181.82 Z0.1`
`...`

I don't see anything wrong. I think it's a problem with your firmware or that you have selected an incorrect Gcode flavor.

kirilledelman, 2018-07-31:
I couldn't replicate this last time I printed. This is really odd. I'll reopen this if it persists.

closed 2018-07-31 by kirilledelman