Ultimaker/Cura #4037: [3.4.0]Printer Prefixes no longer based on printer names, but on printer profile names

desrtfx, 2018-07-06:
I just realized that the prefixes are no longer based on the actual names of the printers, but on the profile names used when adding the printers.

Before 3.4, my _Anet A6_ was abbreviated _AA6_, now it is abbreviated _CFDMP_. My _Ender 3_ abbreviates as _CCR10_ (should be _E3_).

**Application Version**
3.4

**Platform**
Windows 10

**Printer**
N/A because applies to all printers

**Steps to Reproduce**
Add a new printer based on an existing device profile. Choose a different name and check the prefix that will be used when saving gcode.

**Additional Information**
Making the prefixes user configurable would be a great improvement.

barrettford, 2018-07-06:
I second this.  My gcode used to be prefixed with TT_ for my Tevo Tornado, but now they come out with CCR10_

ianpaschal, 2018-07-09:
We've been discussing how to address this and it will likely become a non-issue if/when it's possible to simply specify your own prefix per machine.

desrtfx, 2018-07-09:
Sounds good! Custom prefixes will definitely be the best way to go.

diegopradogesto, 2018-07-10:
We made a poll asking users for their preferences and the results were that users prefer to customise the names. As far as I know we already have an internal ticket for that in our backlog, so still a WIP.

alekseisasin, 2018-07-19:
we already have a ticket for this:
Plese check the ticket #3753

closed 2018-07-19 by alekseisasin

fieldOfView, 2019-10-08:
I'm working on a plugin that will let you customize the print job prefix:
https://community.ultimaker.com/topic/29734-custom-printjob-prefix-plugin/

desrtfx, 2019-10-10:
Thank you very much!

Just installed it on my Cura. Really nice :)