Ultimaker/Cura #4039: Cura occupies COM (serial) port, prevents other applications from using it

frank26080115, 2018-07-07:
**Application Version**
3.3.1

**Platform**
Windows 8.1 64-bit

**Printer**
Ultimaker 2

**Steps to Reproduce**
1. Have Cura running in a window, not doing anything

2. Plug in an Arduino Nano

3. try to use Arduino IDE to upload a sketch

**Actual Results**
Upload fails because avrdude reports that the COM port has access denied because it is in use

Only way to fix is to unplug the USB cable to the Arduino and plug it back in, then reattempt upload

**Expected results**
COM port is available for use and the upload succeeds

**Additional Information**
Please provide an option to turn off whatever automatic COM port detection Cura is using and causing this conflict

smartavionics, 2018-07-07:
You can disable the USB plugin, that's what I have done to stop Cura hoarding the serial ports.

frank26080115, 2018-07-08:
Thank you smartavionics

closed 2018-07-09 by ChrisTerBeke

FalconFour, 2019-02-18:
wtf is this "won't fix" nonsense? Cura can't even control my serial printer - it detects it as COM31 but it won't communicate with it. I still end up having to use Repetier-Host on top of Cura because my printer doesn't have an SD card slot or display. Now, other programs can't use the serial ports and I tracked it down to "cura.exe" claiming every "usbser" instance.

I don't know how to dig through and rip plugins out of a bone stock installation of Cura, but because of this issue, I'm going to have to figure it out.

WTF about this "won't fix" sh*t?

Ghostkeeper, 2019-02-18:
Yeah that was not correct.

I consider this a duplicate of https://github.com/Ultimaker/Cura/issues/70, https://github.com/Ultimaker/Cura/issues/1478, https://github.com/Ultimaker/Cura/issues/5202 and https://github.com/Ultimaker/Cura/issues/5207.

FalconFour, 2019-02-18:
Huge thanks! I don't usually expect comments to result in a resolution so I tend a bit abrasive, haha.

I solved the problem by deleting the USB Print folder from the Program Files\Ultimaker Cura\Plugins folder. I'd really like to see USB printing work in Cura some day (just having started with it, not sure if it worked in the past with basic Marlin FW) but to use it, I'd need to see raw serial logs as well since the printer uses old FW and synchronous commands (won't respond until preheating is complete, throws frequent MAXTEMP alarms at 230C with 225C temp, requiring an M999 and restart).

I just hope this doesn't trend towards "we should just remove USB printing support" :)

Ghostkeeper, 2019-02-18:
We're not about to remove USB printing support, don't worry.

An easier way than to remove the USB printing plug-in in Program Files is to go to the Marketplace, then to "Installed" and uncheck the USB Printing plug-in. Then restart Cura.

FlailAway, 2019-02-28:
> An easier way than to remove the USB printing plug-in in Program Files is to go to the Marketplace, then to "Installed" and uncheck the USB Printing plug-in. Then restart Cura.

Sheeesh, you windows guys! Just having this same issue with Mint 19.01 when I start Marlin I have to close down Cura. How do we Lesser-Linux-Folk :dancing_men:  get to curtail Cura's autocratic nature with the USB.

smartavionics, 2019-02-28:
>  Sheeesh, you windows guys!

Err, careful with your language there!

fieldOfView, 2019-02-28:
@FlailAway, the text you quote explains exactly what you should do, whether you run Cura on Linux, Windows or OSX. Marketplace is a "thing" inside Cura.

FlailAway, 2019-02-28:
> @FlailAway, the text you quote explains exactly what you should do

Thanks FOV, but Ghostkeeper stated
"...in Program Files is to go to the Marketplace,"

Pretty sure a search for "Program Files" will prove to be futile within Mint.

But, I will now look for "Marketplace" within Cura now that you have told me where to look. That darn Clairvoyant's course I took was a waste of money. ;)

FlailAway, 2019-02-28:
OK, just did that and by the way, before posting about an hour-back, I had checked the "Preferences" and other "Settings," and found nothing to do with disable the USB. 

And, now the biggy-#1, {grin} Marketplace is totally empty of anything.
And biggy-#2, the printer PC does not have Internet access. But I just checked on this PC which clearly does have Internet and the Cura 3.6.0 appimage also shows Marketplace also empty.

fieldOfView, 2019-02-28:
Even if the Marketplace can not download the list of plugins available to install, there should still be an "Installed" tab (towards the top right of the Marketplace dialog). That tab lists all plugins that are installed, including plugins that come preinstalled like USB Printing.

FlailAway, 2019-02-28:
> there should still be an "Installed" tab (towards the top right of the Marketplace dialog).

Aaahhh, OK there is but it looks grayed out so I didn't think to try since the entire page was empty. After disabling the USB Printing and a Cura restart, it was all good with Marlin.

After the Cura restart, I checked marketplace again and the "installed" list was open and all visible. 

Might I suggest that this should be the default as looking at a blank page is not conducive to encouraging one to forage around in there. One might assume, as I did, that the blank page meant that nothing was installed yet. I was not aware that Cura was made up of mostly Plugins.

Also on that topic, perhaps an option within the "Preferences" menu dialog for disabling USB might be a nice addition. While I disagree with the concept of USB defaulting to connected, I can see the need to have that as default for the uninitiated.

Ghostkeeper, 2019-03-01:
> Thanks FOV, but Ghostkeeper stated
> "...in Program Files is to go to the Marketplace,"
> 
> Pretty sure a search for "Program Files" will prove to be futile within Mint.

Sorry. I clarified my comment with some brackets now. FalconFour mentioned that he deleted the plug-in in Program Files which is unnecessary. That's what I wanted to explain.

rapzak, 2020-01-31:
Took me a lot of hours to realize it was not my embedded programming there was the issue, im doing project with serial/mass storage on 1 usb, and it fucked up - but relized just now it only does it with CURA open.... and the i found this thread.... hours of wasted time :(

agibson2, 2020-06-29:
I just came across this too.  I couldn't figure out why the software that monitors/controls an RC Charger was not working after installing it.  I was going down the route of USB drivers for the chip that the charger used and eventually figured out that it worked after quiting Cura.  I really did not expect that.  I disabled USB printing in Cura to free up the port.  If someone prints with USB though and also has other COM ports for other uses, Would it still tie up the other COM ports?  Disabling USB printing in Cura would not be an option for them.

I imagine the idea of USB plugin's behavior is to make sure the 3d printer will work even if someone connects it later or turns it on later and it was not responsive before.  I wonder if a comprimise could be to have some kind of behavior maybe where if a com port doesn't respond then maybe close the port and then open it again some time in the future to find printers.  This would allow other apps to take ports that are not meant for 3d printers and still allow Cura to find 3d printers on ports automatically.

Ghostkeeper, 2020-07-02:
Note that this issue is closed in favour of https://github.com/Ultimaker/Cura/issues/5207.

gm310509, 2020-11-30:
Am I missing something or most people missing the elephant in the room? Why is it that the only alternative to Cura locking down every serial port is to completely disable USB printing in Cura? This makes no sense to me - so obviously i (and possibly jorgie0 along with all the people who liked jorgie0's reply) must be missing something that the rest of the world should see.

So, I reiterate, a program that locks resources that it is not using, nor never intends to use is not behaving correctly and thus should be fixed - IMHO.

Ghostkeeper, 2020-12-01:
It's not the only alternative. Please see #5207 and continue discussion there.