Ultimaker/Cura #4041: Material retraction settings are not updated in the Print Setup/Profile

elmerohueso, 2018-07-08:
**Application Version**
3.4

**Platform**
Windows 10

**Printer**
Ender 3

**Steps to Reproduce**
1. Go to Manage Materials
2. Create a new material with Retraction Distance and Retraction Speed
3. Close Material Manager
4. Select newly created material from drop-down
5. Observe that Retraction Distance and Retraction Speed settings are not changed in the print setup/profile

**Actual Results**
Retraction Distance and Retraction Speed settings are being inherited from the printing profile instead of the material profile.

**Expected results**
Retraction Distance and Retraction Speed settings in the printing profile should be overwritten by those in the selected material profile.

**Additional Information**
fdmprinter definition doesn't appear to have any tags referencing the Retraction Distance/Speed of the material, similar to the default_material_temperature tags, so it may not even be set up to inherit retraction settings from the material profile.

fieldOfView, 2018-07-09:
Please attach a (zipped) project file (File -> Save project)

In all likeliness, there are Retrection distance/speed settings in your quality profile. These always override your material profile.

elmerohueso, 2018-07-09:
Here's the project:
[CE3_3DBenchy_-_LEGO_compatible_-_BottomTop_connections.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2177350/CE3_3DBenchy_-_LEGO_compatible_-_BottomTop_connections.curaproject.3mf.zip)

How do I change the profile to inherit the Retraction settings from the material?  Do I have to export it, edit, then re-import?  I see it inherits the temperature settings with "material_bed_temperature = =default_material_bed_temperature" and "material_print_temperature = =default_material_print_temperature", but I don't see similar tags in the fdmprinter definition json to make it inherit retraction settings.

fieldOfView, 2018-07-09:
Go to Help -> Show configuration folder (and quit Cura)
In the "quality" folder, open the file named "fdmextruder_#3_ender3_0.2.inst.cfg" in a text editor
Remove the following lines:
```
retraction_amount = 7
retraction_speed = 60
```
Save the file and start Cura

fieldOfView, 2018-07-10:
The reset button only works until a setting is saved to a profile. A way to remove a setting from a saved profile (quality_changes) is sorely missing. The only way to do so is to edit the inst.cfg file.

smartavionics, 2018-07-10:
>  A way to remove a setting from a saved profile (quality_changes) is sorely missing.

Hi Aldo, isn't this the kind of challenge you enjoy?!

fieldOfView, 2018-07-10:
[image]

If fixing this is not on the roadmap for Cura 4.0, I might have a stab at it.

smartavionics, 2018-07-10:
Good man!

ianpaschal, 2018-07-10:
But what would you be resetting to if not the quality profile? Generic vanilla settings?

fieldOfView, 2018-07-10:
It would "reset" to the next value in the stack, be it material, variant, definition_changes or definition. In most use-cases the material value or the definition value. 

It is not really setting the value to some "default" value, but removing the value from a container in the stack.

smartavionics, 2018-07-10:
My take on this is that there should be a button or menu item that would zap whatever change the user has made and saved into the current profile. The setting would revert to whatever it was before the user made any changes to it.

ianpaschal, 2018-07-10:
> It is not really setting the value to some "default" value, but removing the value from a container in the stack.

Well yes, but I meant "how far back" do you go. What's the baseline. I guess if it's "undo one level" that would be fairly intuitive.

fieldOfView, 2018-07-10:
You know how stacks work, right?

The current "reset" arrow works by removing a setting from the "user" container. This it can only work for settings set in the sidebar that are not yet in a quality-changes container. The quality-changes "profile" needs a similar control to remove a setting from it. Probably not in the sidebar, but in the Profiles pane of preferences.

fieldOfView, 2018-07-10:
The problem is that if a setting is in a readonly container (such as a "quality" profile) it can not be removed from the stack.

ianpaschal, 2018-07-10:
Yeah I get how they work but that's why I'm wondering "reset makes sense that it removed the user changes but if you want to reset further than that, how far? does the reset arrow just stay there so you can keep going all the way back to global?"

stefanbergh, 2019-06-16:
I have the same question and did not find the answer. Can you make the print profile get retraction settings from the material profile, just like temperate? 

Otherwise, whats the point of having it in material profile?

Ghostkeeper, 2019-06-17:
The quality profiles override the material profiles.

There can be multiple quality profiles per material. If a certain quality level requires different retractions (e.g. shorter retractions to print faster at the cost of more stringing) then that quality level will override the retraction distance setting and the one in the material profile will not get used any more. If ALL quality profiles override the retraction distance, I'd say that the quality profiles are set up wrong but indeed there is no point in having it in the material profile then.

stefanbergh, 2019-06-17:
Yes, I get that the print profile should override the material profile but the problem is that there is no way to "reset" the print profile settings to material profile settings. 

Of course no big deal as a can just skip material profiles all together and just make print profiles for material/print settings together. As the settings exists in material settings also I'd guess that there was a plan of using it like I would like but either it hasn't been implemented yet or I just don't understand how it works.

htkoca, 2019-10-10:
There seems to be many issues and threads on the web detailing this confusion of the link between materials and profiles, as well as the inability to revert to material settings for a profile.

https://github.com/Ultimaker/Cura/issues/2039
https://www.reddit.com/r/3Dprinting/comments/5q5emh/q_removing_settings_from_cura_profiles/
https://github.com/Ultimaker/Cura/issues/1135
https://github.com/Ultimaker/Cura/issues/2653

It seems counter-intuitive that printer profiles override materials. And it looks like there are a few people who agree, hence the issues. Since this is profile settings > material settings are indeed intended behaviour of cura, at least allow a way to users to set the value to correspond with the material settings. Right now it's either the `revert` or `fx` button, neither lets users go back to using material values.

jtatum, 2020-07-19:
Can this be reopened? It's still an issue, even in current Cura.

Ghostkeeper, 2020-07-21:
I'll open this back up for discussion. It needs discussion, because the profile system is powerful enough already to allow this, but some profile authors don't fully realise the consequences of their overrides.

So the value in the material profile serves as the default, but the quality profile can override that. To make this concrete, you could have the following hypothetical profiles all specialized for a specific printer, say, the Ender 3:

* Material profile "PLA": Retraction distance set to 5mm.
* Quality profile for PLA "Fast": Retraction distance set to 2mm.
* Quality profile for PLA "Normal": No override.
* Quality profile for PLA "Fine": No override.

So normally for PLA, the retraction distance would be 5mm, but for Fast it's 2mm because Fast needs to be faster and long retractions take time.

This issue comes up in two cases: For some printers, the profile authors didn't know what they were doing and made the profiles "Normal" and "Fine" also override the retraction distance. The original 5mm doesn't have any meaning any more because all quality profiles override it. Second case, the user wants to adjust the material profile's retraction distance but since he's using "Fast", he doesn't see the change; the change only applies to "Normal" and "Fine" because "Fast" overrides it.

In both cases, the user just really wants to get rid of the value from the built-in quality profile. That's possible; he'd need to change the Material Type of the material, and create his own profiles. A bit of work to restore the other values that were overridden by the built-in quality profiles though. What can we do to improve the workflow for this use case?

One thing that's been discussed before here is not on the table: We won't change the material profile to override the quality profile, because that will break thousands of profiles that are already out there, not to mention some key assumptions that Cura's profile system and user workflows currently make. For instance, we'd have to split up the material profiles to be able to have a separate material profile for each quality level in order to be able to represent the same setting values. That just defeats the purpose of the material profiles for most people.

jtatum, 2020-07-21:
One thing that I think really confuses me is that you just can't see that anywhere. When viewing current settings, I can see if my settings differ from profile settings and there's a button to revert that if so. But when I hit that button, I can't tell if I'm reverting to the quality profile or the material profile. Furthermore, the place I'd expect to show this (Settings -> Profiles -> Global settings or extrude) doesn't list when a quality profile includes a retraction setting. It shows a ton of profile settings, but for some reason not that one, making it look like all your profiles are free of retraction settings even if they aren't.

nallath, 2020-07-22:
Since we're discussing it here, here is some documentation wrt the settings & profiles: https://github.com/Ultimaker/Cura/wiki/Profiles-&-Settings

ppriest, 2022-07-03:
As an experienced software eng, I can see the intent in the modelling of the various physical aspects of the printers, and the layering of the config. It's not a perfect abstraction, as it assumes that each of these elements (nozzles, hotends, extruders, frame) are discrete in terms of their impact on the settings (each setting in a given configuration should come from precisely one place) - yet doesn't compartmentalise each of the settings and associate them with one discrete component (e.g. assuming that the temperature is a function only of the material and restricting it as such).
However, with this flexibility comes an issue in presenting it in a UI, and in terms of the user conveying where the setting has come from, and worse, allowing the user to control where it should be persisted to ( short of having them drop into files on the filesystem, worry about how an upgrade to the software would impact their manual changes etc.).

To properly solve this, I think the UI would need to add an advanced settings mode (say a toggle, disabled by default).
When enabled:

- Hovering/Expanding a setting should reveal from what component the setting came from, and any others it may be overriding
- Add the ability to 'pin' a setting to the value coming from one of the components, even though it would normally be overridden - this mitigates the concern of changing the default arbitration order of components.
- Add the ability to override the value coming from a specific component, and save/remove it, rather than the current system where a manual change is layered on top of all the other components

I could mock something up if a UI developer was interested in discussing further...

5axes, 2023-03-07:
My Workaround solution : https://github.com/5axes/ProfilePlus