Ultimaker/Cura #4042: ubuntu: cura 3.3.1 does no longer save files - URGENT!

herrdeh, 2018-07-09:
**Application Version**
3.3.1

**Platform**
ubuntu 18.04

**Steps to Reproduce**
1. open or create file
2. save project...
3. save
4. check in file manager

or:
1. slice project
2. save to external SD
3. check in file manager - find a 0byte file

**Actual Results**
see above

**Additional Information**
When launched from terminal, I see this when saving a file: 
`.
2018-07-09 11:05:14,320 - ERROR - [MainThread] UM.Logger.logException [81]: Exception: Unable to write to file PI3_3-PLA-GH-36-8-oben&unten-m-Verst: zero-size array to reduction operation minimum which has no identity
2018-07-09 11:05:14,324 - ERROR - [MainThread] UM.Logger.logException [85]: Traceback (most recent call last):
2018-07-09 11:05:14,325 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/UM/Qt/Bindings/OutputDeviceManagerProxy.py", line 122, in _writeToDevice
2018-07-09 11:05:14,325 - ERROR - [MainThread] UM.Logger.logException [85]:     device.requestWrite(nodes, file_name, limit_mimetypes, file_handler, **kwargs)
2018-07-09 11:05:14,325 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/uranium/plugins/LocalFileOutputDevice/LocalFileOutputDevice.py", line 131, in requestWrite
2018-07-09 11:05:14,325 - ERROR - [MainThread] UM.Logger.logException [85]:     self.writeStarted.emit(self)
2018-07-09 11:05:14,326 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/UM/Signal.py", line 212, in emit
2018-07-09 11:05:14,326 - ERROR - [MainThread] UM.Logger.logException [85]:     self.__performEmit(*args, **kwargs)
2018-07-09 11:05:14,326 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/UM/Signal.py", line 319, in __performEmit
2018-07-09 11:05:14,326 - ERROR - [MainThread] UM.Logger.logException [85]:     signal.emit(*args, **kwargs)
2018-07-09 11:05:14,327 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/UM/Signal.py", line 212, in emit
2018-07-09 11:05:14,327 - ERROR - [MainThread] UM.Logger.logException [85]:     self.__performEmit(*args, **kwargs)
2018-07-09 11:05:14,327 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/UM/Signal.py", line 315, in __performEmit
2018-07-09 11:05:14,327 - ERROR - [MainThread] UM.Logger.logException [85]:     func(dest, *args, **kwargs)
2018-07-09 11:05:14,328 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/cura/plugins/UFPWriter/UFPWriter.py", line 26, in _createSnapshot
2018-07-09 11:05:14,328 - ERROR - [MainThread] UM.Logger.logException [85]:     self._snapshot = Snapshot.snapshot(width = 300, height = 300)
2018-07-09 11:05:14,328 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/cura/Snapshot.py", line 92, in snapshot
2018-07-09 11:05:14,328 - ERROR - [MainThread] UM.Logger.logException [85]:     min_x, max_x, min_y, max_y = Snapshot.getImageBoundaries(pixel_output)
2018-07-09 11:05:14,329 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/cura/Snapshot.py", line 31, in getImageBoundaries
2018-07-09 11:05:14,329 - ERROR - [MainThread] UM.Logger.logException [85]:     min_y, min_x, min_a_ = numpy.amin(nonzero_pixels, axis=1)
2018-07-09 11:05:14,329 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/numpy/core/fromnumeric.py", line 2372, in amin
2018-07-09 11:05:14,329 - ERROR - [MainThread] UM.Logger.logException [85]:     out=out, **kwargs)
2018-07-09 11:05:14,329 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/numpy/core/_methods.py", line 29, in _amin
2018-07-09 11:05:14,330 - ERROR - [MainThread] UM.Logger.logException [85]:     return umr_minimum(a, axis, None, out, keepdims)
2018-07-09 11:05:14,330 - ERROR - [MainThread] UM.Logger.logException [85]: ValueError: zero-size array to reduction operation minimum which has no identity
`

This is in terminal when saving gcode to an external SD:

`2018-07-09 11:08:22,320 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [611]: Slicing took 9.094295501708984 seconds
2018-07-09 11:08:22,321 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [612]: Number of models per buildplate: {0: 13}
2018-07-09 11:08:22,321 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [626]: See if there is more to slice...
2018-07-09 11:08:22,321 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [181]: Socket state changed to Closing
2018-07-09 11:08:22,322 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [203]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2018-07-09 11:08:22,322 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [183]: Socket state changed to Closed
2018-07-09 11:08:22,322 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [765]: Backend quit with return code 0. Resetting process and socket.
2018-07-09 11:08:25,085 - DEBUG - [MainThread] RemovableDriveOutputDevice.RemovableDriveOutputDevice.requestWrite [83]: Writing to /media/wd/6C00-EA2B/PI3_3-PLA-GH-36-8-oben&unten-m-Verst.gcode
2018-07-09 11:08:25,086 - DEBUG - [MainThread] UFPWriter.UFPWriter._createSnapshot [25]: Creating thumbnail image...
2018-07-09 11:08:25,094 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/cura/resources/shaders/overhang.shader]...
2018-07-09 11:08:25,328 - ERROR - [MainThread] UM.Logger.logException [81]: Exception: Unable to write to file PI3_3-PLA-GH-36-8-oben&unten-m-Verst: zero-size array to reduction operation minimum which has no identity
2018-07-09 11:08:25,329 - ERROR - [MainThread] UM.Logger.logException [85]: Traceback (most recent call last):
2018-07-09 11:08:25,329 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/UM/Qt/Bindings/OutputDeviceManagerProxy.py", line 122, in _writeToDevice
2018-07-09 11:08:25,329 - ERROR - [MainThread] UM.Logger.logException [85]:     device.requestWrite(nodes, file_name, limit_mimetypes, file_handler, **kwargs)
2018-07-09 11:08:25,329 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDriveOutputDevice.py", line 97, in requestWrite
2018-07-09 11:08:25,330 - ERROR - [MainThread] UM.Logger.logException [85]:     self.writeStarted.emit(self)
2018-07-09 11:08:25,330 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/UM/Signal.py", line 212, in emit
2018-07-09 11:08:25,330 - ERROR - [MainThread] UM.Logger.logException [85]:     self.__performEmit(*args, **kwargs)
2018-07-09 11:08:25,331 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/UM/Signal.py", line 319, in __performEmit
2018-07-09 11:08:25,331 - ERROR - [MainThread] UM.Logger.logException [85]:     signal.emit(*args, **kwargs)
2018-07-09 11:08:25,331 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/UM/Signal.py", line 212, in emit
2018-07-09 11:08:25,331 - ERROR - [MainThread] UM.Logger.logException [85]:     self.__performEmit(*args, **kwargs)
2018-07-09 11:08:25,332 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/UM/Signal.py", line 315, in __performEmit
2018-07-09 11:08:25,332 - ERROR - [MainThread] UM.Logger.logException [85]:     func(dest, *args, **kwargs)
2018-07-09 11:08:25,332 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/cura/plugins/UFPWriter/UFPWriter.py", line 26, in _createSnapshot
2018-07-09 11:08:25,333 - ERROR - [MainThread] UM.Logger.logException [85]:     self._snapshot = Snapshot.snapshot(width = 300, height = 300)
2018-07-09 11:08:25,333 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/cura/Snapshot.py", line 92, in snapshot
2018-07-09 11:08:25,333 - ERROR - [MainThread] UM.Logger.logException [85]:     min_x, max_x, min_y, max_y = Snapshot.getImageBoundaries(pixel_output)
2018-07-09 11:08:25,333 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/cura/Snapshot.py", line 31, in getImageBoundaries
[... cut ...]
2018-07-09 11:08:25,334 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/lib/python3/dist-packages/numpy/core/_methods.py", line 29, in _amin
2018-07-09 11:08:25,335 - ERROR - [MainThread] UM.Logger.logException [85]:     return umr_minimum(a, axis, None, out, keepdims)
2018-07-09 11:08:25,335 - ERROR - [MainThread] UM.Logger.logException [85]: ValueError: zero-size array to reduction operation minimum which has no identity
2018-07-09 11:08:55,045 - DEBUG - [MainThread] UM.Message._onInactivityTriggered`

alekseisasin, 2018-07-09:
HI @herrdeh, I am using Ubuntu and it worked for me fine. Actually, the latest Cura version is 3.4, you can try it.

herrdeh, 2018-07-09:
Thanks for the quick answer.
I'm using this ppa:

> http://ppa.launchpad.net/thopiekar/cura/ubuntu

Which one should I use for 3.4?

Cheers,
Wolf

ianpaschal, 2018-07-09:
The PPA isn't actually maintained by us. Paging @thopiekar

herrdeh, 2018-07-09:
Thanks for the hints. I wrote a notice to thopiekar, the appimage works nicely.
Great support!

Cheers,
Wolf

thopiekar, 2018-07-09:
Hey @ianpaschal, thank you for redirecting this issue here.
Could someone of you Ultimaker guys format @herrdeh 's log here, so it is better readable?

Thanks!

ianpaschal, 2018-07-09:
Turrdurrrr!

thopiekar, 2018-07-09:
A-a-awesome! 🙏 Will take a look onto it in the following days.

@herrdeh Please make sure you have got all updates installed.
Some people only update the Cura package, but you need to update all its dependencies, like Uranium, as well!

nehu92, 2018-08-01:
have the same issue with the same log output with version 4.3.1 in windows 10.

fieldOfView, 2018-08-01:
Why is the UFPWriter trying to create a thumbnail in the first place? Unless you have an Ultimaker S5, Cura should not be writing UFP files.

ChrisTerBeke, 2018-08-01:
@fieldOfView Cura connect will also show these images on UM3.

fieldOfView, 2018-08-01:
@nehu92 is also affected by this issue, and in https://github.com/Ultimaker/Cura/issues/4168 he says he's using a custom printer. @herrdeh, what printer do you use?

herrdeh, 2018-08-01:
Anet A8.
But as mentioned above - appimage 3.4 launched once, and the problem was gone and remained. For whatever reason.

fieldOfView, 2018-08-01:
>  For whatever reason.

That's always scary.

For some reason, Cura remembered (incorrectly?) that you wanted to save .UFP files instead of .gcode files. Also for some reason, creating a UFP file fails. Also it seems yours is not a solitary case, so this may need fixing.

nehu92, 2018-08-02:
In a Facebook group of 3d print several members have the same problem. I dont think this is an isolated case. Why the issue is close? Using the appimage is the solution? I use windows :(

ianpaschal, 2018-08-03:
The original poster closed it. I'll reopen it though.

alekseisasin, 2018-08-20:
@herrdeh , I still cannot reproduce it. Could you please share the whole log file, then I will check all the steps.

herrdeh, 2018-08-20:
as described [here](#issuecomment-403688311), the problem no longer occurred since I once launched appimage v3.4.0. Sorry, I can no longer help you.

Ghostkeeper, 2018-09-21:
Let's assume this was inadvertently fixed in version 3.4.0, then.

closed 2018-09-21 by Ghostkeeper