Ultimaker/Cura #4046: 3.4.0 support first layer issue when using brim as plate adhesion

maxipalay, 2018-07-09:
**Application Version**
3.4.0

**Platform**
macOS

**Printer**
Deltabot

**Steps to Reproduce**
Use brim as plate adhesion and supports touching build plate. Try to slice a hollow cylinder (vertical position) with one end covered (see pictures attached), in such position that support is needed inside the cylinder to hold the covered end of the cylinder (which will print on top).

**Actual Results**
The first layer won't print support inside the cylinder, but the second will. This doesn't let the filament of the support stick correctly to the plate.

**Expected results**
First layer prints support.

**Additional Information**
Please see captures attached.

See how on second layer the support will be printed inside the cylinder but not on the first.

On the outside of the cylinder the support is printed since first layer.

When I say printed I mean what would happen if you were to print this, although it is a slicer issue.

[image]
[image]
[image]

ianpaschal, 2018-07-10:
hmm indeed so it doesn't generate "inside" shapes...

diegopradogesto, 2018-07-10:
This is a duplicate of #3254. This issue is indeed fixed in our current board.

closed 2018-07-10 by diegopradogesto