Ultimaker/Cura #4051: support&infill object interference

herrdeh, 2018-07-10:
**Application Version**
3.4.1
**Platform**
ubuntu 18.04

**Steps to Reproduce**
1. import a flat box, mark as "print as support structure", place it underneath/into your target object
2. import a cylinder, mark as "overlap with other models", set infill density very high, place into the target object - same place as support object
3. slice
4. remove support object
5. slice again

**Actual Results**
The overlap object will print to the very bottom of the build plate.

**Expected results**
The overlap object should only print where intersecting with the target object.

**Additional Information**

As it should be:
[image]
[image]
[image]

As it comes out with additional support object:
[image]
[image]

Example file:
https://www.dropbox.com/s/h63mp6wz18mlkn6/PI3_PLA-GH-unten-36-10-3-m-Verst.curaproject.3mf?dl=0

herrdeh, 2018-07-10:
The problem does not occur with support in general activated.

diegopradogesto, 2018-07-13:
Indeed, this is a bug. Thanks for providing all the information.

As a tip for this model, I would use better "Modify settings for infill of other models" for the cylinders instead of "Modify settings for overlap with other models". In this case, the outer surface of your main model doesn't get affected by the cylinders. See attached picture:
Left cylinder was choose to "Modify settings for overlap with other models" (blue areas shows some artifacts in the model surface)
Right cylinder was choose to "Modify settings for infill of other models" (nor artifacts in the surface)
[image]

Curiously this will serve as a workaround for your problem. Give it a try!

diegopradogesto, 2018-07-16:
For devs.: CURA-5579

diegopradogesto, 2018-07-18:
This is now fixed on master branch. It will be available in next release.

closed 2018-07-18 by diegopradogesto

herrdeh, 2018-07-19:
Thank you very much for taking care for this issue.
Actually, I saw the artifacts mentioned in #3 - good to have a solution for it!

Cheers,
Wolf