Ultimaker/Cura #4052: Bridge walls on layers 2+3 of a bridge.

jarrodasmith, 2018-07-10:
**Application Version**

3.4

**Platform**

All Platforms

**Printer**

CR-10

**Steps to Reproduce**

Select "Enable Bridge Settings" under Experimental.  Customize any of the bridge wall options (e.g. Bridge Wall Speed and Bridge Fan Speed), then select "Bridge Has Multiple Layers".

**Actual Results**

Only skin is treated with bridge settings on the second and third layer of the bridge.  Walls are treated as regular walls on layers 2 and 3 of the bridge.  This most commonly causes the edges of a bridge to sag.

**Expected results**

When you enable "Bridge Has Multiple Layers", walls should be treated as "bridge walls" on layers 2 and 3 of a bridge.  It may or may not be needed to add additional settings to customize bridge wall settings specifically for layers 2 and 3 of the bridge as you have for skin.  Inheriting the wall settings from layer 1 of the bridge may be sufficient.

smartavionics, 2018-07-10:
Hi, thanks for your report. Could you please post images showing how the 2nd/3rd bridge layer walls are failing. Personally, I have not had any issues when the 2nd/3rd layer walls are printed using their normal settings. Thanks.

jarrodasmith, 2018-07-10:
[image]
[image]

jarrodasmith, 2018-07-10:
Granted that is not the best bridge (I've got settings that do better) but clearly illustrates the issue - the center of the bridge is pretty level, but the sides sagged.  You can watch this during the print.  When the nozzle does the perimeter on the second layer of the bridge, it softens the first layers and they sag.  When it does the center (skin) where I slow things down and turn up the fan, it behaves nicely.   If I slow the whole print down and turn the fan up it behaves better but of course then you lose the benefit of custom settings for bridging...you can just print the whole thing like a bridge and it will take forever :-).  Also I print a lot of things that are 100% walls, and I feel confident that the ability to build up a few layers of wall by slowing down and turning up the fan would greatly improve those bridges.

smartavionics, 2018-07-10:
OK, thanks for the photos. Could you please do an experiment for me? Print that same model but stop the print after the 3 bridge layers have been printed and provide a photo similar to the above. Thanks.

Interesting that you turn up the fan for the 2nd and 3rd layer skins as I get better results with no fan at all for those skins.

jarrodasmith, 2018-07-10:
Here's a better "true" example.  This is a 100mm bridge that came out quite decent, but if you turn it over you see that from the side it actually looks much worse than it really is.  There is a "ledge" running along each edge that's about 1mm deep, where the perimeters slightly sagged as the nozzle came back over the first bridge layer at normal print speed/fan settings.  It gets *much* worse the more perimeters you have set.  This example was setup to print with just two perimeters.

[image]

smartavionics, 2018-07-10:
Sorry, that picture is hard for me to interpret. I can't really see what's happening there.

I find it very difficult to get good images of prints.

jarrodasmith, 2018-07-10:
I want to point out a couple things that might explain the different results we are getting:  I'm using a 0.8mm nozzle with 0.4mm layers.  I'm also using a volcano-style nozzle, i.e. short and wide.  My theory is that the edge of the nozzle comes quite close to the previously printed perimeters and that is what causes the sag.  Turning up the fan mitigates this.  I wonder if I was using a more "pointy" nozzle if this issue would be less pronounced.

I'll gladly do your test.  Stay tuned.

smartavionics, 2018-07-10:
Ah, OK, maybe you have a good point there as I am only using a 0.4mm nozzle.

BTW, the reason I use no fan for the 2nd and 3rd skins is because I found that when the nozzle is moving slowly (which I believe is required for those skins to get good adhesion) the fan just blows the air heated by the nozzle directly onto the 1st skin which makes the skin lines sag. By not having any fan, I get less sag on the skins. YMMV.

jarrodasmith, 2018-07-10:
[image]

smartavionics, 2018-07-10:
BTW, can you either build Cura from source or run a Linux version? If so, then I could look into making a version that you could test. Sorry, I can only build for Linux.

jarrodasmith, 2018-07-10:
I think I used about 1/2kg of this yellow filament printing this model in about 100 different combinations of these settings so I'm definitely familiar with what you mean :-)  On my setup, an in-between setting for what I'll call "upper bridge skin" is best.  So I'll do 60% fan for the overall print so it can go pretty fast ~45-50mm/sec, 100% fan for the first bridge layer (10mm/sec), and something like 80% for bridge layers 2/3 so I get decent adhesion but minimize sagging of previous bridge layer(s) as the nozzle passes by.  Always a balancing act :-)

jarrodasmith, 2018-07-10:
In theory I can build for Linux.  Currently in between machines so at this very moment I could build it at work but could not run it at home :-D

jarrodasmith, 2018-07-16:
So I tried to get Cura to build on OSX per the directions and failed due to broken protobuf python dependencies (even inside a virtualenv stack, even with everything installed exactly per the official instructions).  So....I'll be moving ahead with Linux on my brand new Win10 box.  But I'll be on vacation until next week.  Just wanted to let you know I'm still thinking about this.  Thanks!

Ghostkeeper, 2018-08-28:
I wonder if they were not marked as "bridge walls" on the higher bridge layers on purpose?

closed 2020-06-20 by no-response[bot]

Kaptn225, 2022-08-12:
Issue is still present in Cura 5.1.0

Ticking the "bridge has multiple layers" checkbox has no effect, nor do any of the settings that appear. Second and third bridge layers just get printed with regular wall settings. 

Only the first bridge layer can be changed.

https://imgur.com/a/5xpO2TH