Ultimaker/Cura #4053: Minimum Infill Area issue

LemaruX, 2018-07-10:
**Application Version**

**Platform**

3.4.1

**Printer**

Any

**Steps to Reproduce**

Open the and STL file (the example I used is attached, but also occurs with others). Specify a "Minimum Infill Area" (in my example I used 50mm^2). 
[Vampire HDMI Mount.zip](https://github.com/Ultimaker/Cura/files/2181532/Vampire.HDMI.Mount.zip)

**Actual Results**

Some of the smaller top surface areas dissapear when the "Minimum Infill Area" is specified (as shown in attached screenshot)
[image]

**Expected results**

The areas smaller then the specified area should be printed with skin, and the top surface layers should be left intact. (As shown in attached screen shot of how it is without minimum infill area set)
[image]

**Additional Information**

I did mention this on the Ultimaker forum, as can be seen here.  https://community.ultimaker.com/topic/23131-minimum-infill-area/ and it was admitted that this feature is broken. However it was also mentioned that this feature would not perform as I expect it to which goes against the description of this feature on the Ultimaker website which explains that this feature does exactly what I expect. Whilst the feature appears to work for the most part, the issue with the missing top skin remains. More screenshots and description can be found on the Ultimaker forum link,

jackha, 2018-07-11:
@LemaruX Thanks for the find and your report. We'll fix it.

Devs: CURA-5560

Ghostkeeper, 2018-08-24:
This has been fixed now for version 3.5. Good find!

closed 2018-08-24 by Ghostkeeper

ChrisTerBeke, 2018-09-26:
The PR connected to this in CuraEngine is still open...

diegopradogesto, 2018-10-02:
The PR is now merged in master and it seems to work as expected. However it still needs to be tested by our QA team.

Sadly, this won't be fixed in 3.5. We apologize for the extra delay.

diegopradogesto, 2018-10-16:
Nope, we didn't pick anything from master. We just fixed some bugs directly in 3.5. This will be available in 3.6. Hopefully it will be available soon.

michaelkress, 2018-11-09:
Wanted to test this mainly in order to prevent excessive shaking of the printer. In Slic3r this is so useful, I'm looking forward to have this feature working in cura.

LemaruX, 2018-11-14:
Did the fix make it into 3.6?

dschraitle, 2020-02-02:
I think I'm experiencing something related to this issue in Cura 4.4.1. When my Minimum Infill Area is set too high, Cura will not print a bottom layer, but instead start printing infill right away.

In this screenshot you can see that some characters are large enough to get skin, but others get infill directly on them, even though there is just empty space below.
[image]

Setting Minimum Infill Area to 0 makes it so that everything gets a proper bottom layer.

Layer Height is .28mm
Bottom Thickness is .8mm
Bottom Layers is 3

Ghostkeeper, 2020-02-04:
A similar problem existed in 4.4, which is fixed in 4.5. See also https://github.com/Ultimaker/Cura/issues/7032.