Ultimaker/Cura #4054: Generation of Support Matieral

FynDestiny, 2018-07-10:
**Application Version**
All

**Platform**
Windows

**Printer**
Creality CR-10S
[CCR10_Stark-Lid.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2181609/CCR10_Stark-Lid.curaproject.3mf.zip)

**Steps to Reproduce**
Doesn't really matter.

**Actual Results**
Many of the support designs have "floating" elements. In the attached zip file you should see what I mean. Look at layer 5, then layer 6, you will see elements without support for themselves.

**Expected results**
I would expect all elements of the support material to be supported from the baseplate up. Granted, not all Support Patterns are appropriate for the item in question (I will attach a zip file with the STL, it will be easier for you to see for yourself), but irrespective of the intended item to print, all supported areas should start from the base (or appropriate surface of the item).

**Additional Information**
[Stark-Lid.zip](https://github.com/Ultimaker/Cura/files/2181680/Stark-Lid.zip)

I hope you can understand what I mean from the above information, if not, please email me.

ianpaschal, 2018-07-16:
I looked through your project file and it appears to be working perfectly.

Which is why...

>*Steps to Reproduce*
>Doesn't really matter.

...it totally does. We don't ask that question just because we like to make people type.

The results from layer 5 to 6 are to be expected, it's simply the beginning of support structures starting to form on top of the infill (all are touching the infill, none are actually floating).

Have you actually tried printing and getting bad results?

FynDestiny, 2018-07-17:
Hi
The only reason they are not floating is from increasing the support density, which is quite a waste of material. Leaving it set to default, you'll find several "floating" bits then.

The ideal support pattern for the lid is concentric, as the lids round, but even that pattern has flaws in it.
The first layer looks fine, you wouldn't expect any problems looking at it, but then the second layer tries to print material where there is none on the first layer, not even a slight connection, and again on other layers onwards up the layering. Which doesn't seem at all correct.

Whenever I slice a design, I always go through the layers to see if it needs support, then again with support to make sure everything is supported.

It would be nice if the standard support patterns, could have additional support added by the user, this would alleviate any possible issues.
I tried the add-on that was written for custom support, but, personally, I think it's crap.

ianpaschal, 2018-07-18:
Unless you're actually getting bad print results from this we're not going to make a re-write to how that is generated. If every square micrometer needed something under it would be impossible to print any object that was not vertical walls/lines. This is a reality of the manufacturing process.

>It would be nice if the standard support patterns, could have additional support added by the user, this would alleviate any possible issues. I tried the add-on that was written for custom support, but, personally, I think it's crap.

This is a long requested feature and incredibly difficult to implement. We have that plugin graciously contributed by someone in their free time, as well as our own implementation in the works. Either way, though, neither is intended for what you're trying to do because it's not necessary.

I will make a sticky note of this issue and put it on our "to decide how we're going to handle it" board but without any evidence that it's negatively affecting the print, I can pretty much guarantee we won't be able to make it a priority.

FynDestiny, 2018-07-18:
That's OK.
Then I apologize to the person who wrote the customizable support.

Though, right now, the only thing that I would really appreciate being considered, is that the current standard support patterns, for example the concentric, are re-coded so that they at least support themselves, instead of printing support material in mid air.

Thank you for taking the time to respond to my rather, rude and ill mannered messages. I have realized this morning, this has been due to me forgetting to reorder some medication. My apologies.

ianpaschal, 2018-07-23:
We discussed this as a team and unfortunately we don't feel it's an issue enough to spend time changing things.

There will ALWAYS be unsupported areas in a 3D print unless you only print straight up and down surfaces, or use a 100% solid support structure.

But the grid support works pretty well, and the concentric one already better. You can see that in the picture I've attached. Everything is crossing support lines and well supported.

[image]

So yeah, as a team, we really feel there's nothing to fix here. Maybe if you were getting some really bad failed prints, but we have yet to see that so as far as we are concerned, there are no bugs here and we can't really spend time fiddling with it for negligible benefits and the possibility of messing up something else. If it ain't broke, don't fix it.

closed 2018-07-23 by ianpaschal