Ultimaker/Cura #4055: Not process start g-code including some parameter and Not display layer view

joonghochoe, 2018-07-11:
**Application Version**
Cura master version

**Platform**
Windows 10 (Notebook : 4G, Desktop : 8G Memory)
Notebook : Core i5, Intel HD Graphics 4000, OpenGL 4 compatible mode(OpenGL 2)

**Printer**
1. All printers that use parameters(ex:{machine_nozzle_size}) in Start G-code(ex: IMADED3D JellyBOX)
2. All printers

**Steps to Reproduce**
1. Slice -> Save G-code -> View G-code
2. Slice -> change simulation view

**Actual Results**
1. not process start g-code including parameter(ex:{machine_nozzle_size})
2. not display simulation view

**Expected results**

**Additional Information**
1. if I change get_value of class GcodeStartEndFormatter(Formatter) to def get_value(self, key, args, kwargs)..., process start g-code
2. when change solid view to simulation view
UM.Logger.logException [84]: Exception: An exception occurred while creating layer mesh.
UM.Logger.logException [88]: Traceback (most recent call last):
UM.Logger.logException [88]:   File "C:\stellamove\cura-build\build\inst\lib\cura\plugins\SimulationView\SimulationView.py", line 609, in run
UM.Logger.logException [88]:     layer = layer_data.getLayer(layer_number).createMesh()
UM.Logger.logException [88]: AttributeError: 'NoneType' object has no attribute 'createMesh'
20

joonghochoe, 2018-07-27:
The second problem was solved. But the first problem seems to remain. 3D printers that use variables(ex, {layer_height}, {infill_sparse_density}...) in start gcode all have this 1st problem.

Ghostkeeper, 2018-09-14:
I'm not able to reproduce the first problem (any more). This is the output of the g-code:

    ;FLAVOR:Marlin
    ;TIME:601
    ;Filament used: 0.218361m
    ;Layer height: 0.1
    ;Generated with Cura_SteamEngine master
    M82 ;absolute extrusion mode
    ;---------------------------------------
    ; ; ; Jellybox Start Script Begin ; ; ;
    ;_______________________________________
    ; M92 E140 ;optionally adjust steps per mm for your filament

    ;              Print Settings Summary
    ; (leave these alone: this is only a list of the slicing settings)
    ; (overwriting these values will NOT change your printer's behavior)
    ;             sliced for :  IMADE3D JellyBOX
    ;        nozzle diameter :  0.4
    ;      filament diameter :  {material_diameter}
    ;           layer height :  0.1
    ;       1st layer height :  0.3
    ;             line width :  0.4
    ;  outer wall wipe dist. :  0.2
    ;      infill line width :  0.4
    ;         wall thickness :  0.8
    ;          top thickness :  0.8
    ;       bottom thickness :  0.8
    ;         infill density :  20
    ;         infill pattern :  grid
    ;      print temperature :  210
    ;  1st layer print temp. :  210
    ; heated bed temperature :  55
    ;    1st layer bed temp. :  55
    ;      regular fan speed :  100.0
    ;          max fan speed :  100.0
    ;     retraction amount  :  6.5
    ;   retr. retract speed  :  25
    ;     retr. prime speed  :  25
    ;   build plate adhesion :  skirt
    ;                support ?  False
    ;             spiralized ?  False

The settings are filled in properly with the global settings.
Only the material_diameter does not get replaced because there is no value for it in the global settings. We defined that as a setting specific to an extruder train. You'd have to get that from an extruder train by getting `{material_diameter, 0}` or something like that.

Layer view looks fine and gives no errors either.

joonghochoe, 2018-09-14:
I confirmed that it was modified in the master version. Thank you.

closed 2018-11-14 by Ghostkeeper