Ultimaker/Cura #4058: Ability to set a custom temperature for ironing

Chompworks, 2018-07-11:
When ironing, some filaments behave better when the extrusion temperature is higher than the typical extrusion temperature for shell walls/infill etc. This may be down to increased viscosity, or even just the nozzle heating the previous layer more.

Would it be possible to add/expose a temperature setting for ironing, so that ironing passes can use their own temperature setting?

Functionality would be:
_normal printing_
Reach ironing section of gcode
Set extruder to ironing temperature
Begin ironing travel passes
...
Finish ironing travel passes
Set extruder to standard temperature for extrusion
_resume normal printing_

Default value for this new field in Cura would be to use the standard material setting.

alekseisasin, 2018-07-13:
Your feature request has a couple of bottlenecks:
 1. If you need a different temperature for ironing then it means during the printing the nozzle should constantly update its temperature (ironing: 220 Cº, cooling down to 200 Cº, print and again for the next layer). In fact, it will increase the printing time
2. If you need special ironing for the last layer. It will work if your layer is flat (then you can apply post processing to set temperature for the last layer), if not then again the last layers will need to be printed with different temperatures.

Chompworks, 2018-07-13:
to me, thats fine.

For point 1: Ironing already adds to the printing time, and in the pursuit of smooth top layers and reducing post-processing, what does waiting a little while longer for nozzle heating/cooling matter in terms of a print that may be 2 hours, 5 hours, 10 hours, or even longer? ;)
Plus, if this value was exposed, it would only take additional time if the ironing temperature value was changed. Otherwise the ironing functionality would work exactly the same as it currently does, with the value defaulting to the standard extrusion temperature, and no delay due to temp changes.

For point 2: for me I have a lot of models where ironing may happen at layer 10, layer 15, layer 45, layer 82 all in the same print. So i think that point is ultimately very model dependent. I could hand tweak gcode, but that is prone to human error in so many ways.
To me it makes more sense to allow the flexibility to choose by exposing the temperature value, even if its hidden away in the experimental section for a few cura releases?

alekseisasin, 2018-07-13:
We do not have hidden settings. What you can try is to change the temperature for specific layers via PostProcessing Plugin. This plugin helps adjust prepared g-code for special scenarios, like yours. Also, you can add own script to this plugin.

Chompworks, 2018-07-13:
I think you may have misunderstood my mention of "hidden" - please don't take it the wrong way.
Some settings are visible or hidden in Cura depending on whether a user goes into settings visibility and ticks or unticks them - this is what I meant by hidden.

Sadly the proposed workaround doesn't work well. For example, when using coloured filaments layer height and extrusion temperature can make a big difference in terms of how the layer looks, and can change the outward colour appearance. So using the existing post processing script to tweak the entire layer for the benefit of ironing introduces a different problem to solve another.

As for writing my own script - trust me, my code is not disciplined enough for anyone to want that outcome! ;)

alekseisasin, 2018-07-16:
It is ok, :), I discussed your request with colleagues and unfortunately, we cannot add the setting to the Cura by default. The reason that some printers might have restriction how frequent the print temperature might be updated, if printer tries to update the temperature every 3 or 10 sec (different printers might have own range) it might damage hotend / temperature controllers
The only way for now is to make the PostProcessing script. You can make a pull request with your code , we and other github contributers can give advice how to improve it or fix. (Tip: in script while parcing a gcode you need to find a layer which does not change the height/ printing volume and by fact it is ironing)

Ghostkeeper, 2018-08-27:
We have a guideline here that says that if you need to change the temperature for less than 50 seconds, you might as well not change it at all. This is because frequent changes to the target temperature cause the PID regulator of the temperature to freak out and produce wildly inaccurate temperatures.

kyjak, 2019-01-11:
I vote for this feature. It is really needed, the materials might have a very different GOOD printing and a very different GOOD ironing temperature. The time is not an issue, and we are not talking a big difference, between these. At least the very top layer could have this setting.

curvi, 2019-06-25:
I would want this too, Kinda. I have the problem that I have a VERY potent part cooling fan installed. It works great and since it's tuneable I can do almost any overhang without support. However, since I have it my ironing is doing more harm than good. I noticed, that when I manually turn off or lower the part cooling fan during that time I can again get those fine smooth top layers! Please could you at least give us an option for part cooling during ironing!? - After all, cooling+ironing sounds a little ironic :D.

dhscord, 2019-06-27:
> We have a guideline here that says that if you need to change the temperature for less than 50 seconds, you might as well not change it at all. This is because frequent changes to the target [...]

If you are ironing something for less than 50 seconds then I'm not sure what you are trying to achieve. 50 seconds of ironing isn't very much surface area across a layer. Maybe have some threshold for the feature like a minimum time for this feature? Anyway, as a potential experimental feature I guess some sort of caveat would be acceptable.

Ghostkeeper, 2019-06-28:
> If you are ironing something for less than 50 seconds then I'm not sure what you are trying to achieve. 50 seconds of ironing isn't very much surface area across a layer.

With applications as popular as Cura, the question of "who would want that?" is not really a problem, usually :stuck_out_tongue: 

Crude example:
[image]

The part that the user really wanted to have ironed is the flat area in the middle. But the slope is so shallow that each of the layers get a thin strip of ironing which takes less than 50 seconds to iron.

dhscord, 2019-06-28:
Interesting example, I wasn't aware that Cura would attempt to iron a layer that was actually part of a gradient.

However, I kind of think this has proved my point - he was ironing something for less than 50 seconds my mistake rather than because he wanted to. Having a minimum ironing threshold of xx seconds before Cura allows the temperature change would have been fine in this case. He wouldn't have cared anyway since really he didn't want to iron these bits.

kyjak, 2019-06-28:
What about a setting "ironing threshold" in seconds? CURA would iron only surfaces that take X seconds to iron. The setting could be in other dimension, not time, for example mm2.

dhscord, 2019-06-28:
> What about a setting "ironing threshold" in seconds? CURA would iron only surfaces that take X seconds to iron. The setting could be in other dimension, not time, for example mm2.

This feels like a different requirement but, for the sake of discussion. It sounds like you are suggesting something like this?

`Enable Ironing -> Minimum ironing area in mm2 (only performs ironing on a particular layer if the surface area to be ironed for that layer exceeds the area specified)  -> Ironing temperature (set a printing temperature to be used when ironing)`

Given this statement further up the thread...
> we cannot add the setting to the Cura by default. The reason that some printers might have restriction how frequent the print temperature might be updated, if printer tries to update the temperature [...]
... this would suggest ironing area is not an ideal measure since the issue is time between temperature changes.

If you instead had the link as follows... 

`Enable Ironing -> Minimum ironing time (only performs ironing on a layer if the calculated ironing time in seconds for that layer is greater than the number specified) -> Ironing temperature (only available where minimum ironing time is greater than 50 seconds)`

... then that would theoretically get over the time issue. The key is that the ironing temperature would not be available unless the higher level time was > some safe number?

However, as I mentioned, that seems like a different requirement. My original thought was...

`Enable Ironing -> Ironing temperature -> Minimum ironing time for temp change (only attempts to change the printing temperature for the ironing part of a layer if that layer's ironing time is greater than the number of seconds specified. This is to stop printer issues caused by rapid attempts to change the target temperature) [minimum = 50] `

This seems simpler.

I suppose having a way to specify a minimum ironing time or ironing area (the top two) is likely an in-application setting (since you are deciding whether to write specific ironing GCODE or not) whereas having a way to set a minimum time to use a temp change could be achieved in post-processing by comparing the timestamps in the GCODE either side of an ironing layer.

I just wish I had the skills to write the post-processing code :)

kyjak, 2019-06-28:
I think we haggle here about nothing. My view is, if users own machines that could handle 5 temp. changes per print and this functionality is desired. Why not do this? Is it really an issue if 100% of world printers are not compatible with a new change/setting in Cura? 
For me, it does not matter much. People eventually will find out how to use the feature. Or make a workaround with a script to do so. This is applicable to a lot of feature requests here..

dhscord, 2019-06-28:
This...
'if printer tries to update the temperature every 3 or 10 sec (different printers might have own range) it might damage hotend / temperature controllers'

kyjak, 2019-06-28:
> 
> 
> This...
> 'if printer tries to update the temperature every 3 or 10 sec (different printers might have own range) it might damage hotend / temperature controllers'

what about this?

dhscord, 2019-06-28:
You will struggle to get native support for a feature that may damage the user's printer. Assuming that the statement above is accurate.

Anyway, I would also love to have this feature. But I see the problem with implementing it natively.

Ghostkeeper, 2019-07-01:
I don't think it would actually damage any temperature controllers, but they do tend to go way outta whack. Initial tests for dual extrusion printing with stand-by temperatures had this problem, that if you had too many temperature commands in a short amount of time it would veer off by 100 degrees or so, even if the actual temperature changes were in the range of 10 degrees.

dhscord, 2019-07-01:
Ok, so I'm gonna have to learn Python then :)

Sophist-UK, 2020-04-07:
My experience is that increasing the temperature by (say) 5c for ironing helps both to get a good slow flow of filament and to melt the existing surface. I cannot speak for other printers, but my head heats up at c. 2c per second. It cools at a slower rate when NOT extruding - but I would imagine it would cool faster when it was extruding.

So, what I would want to see for my printer would be to start to increase the temperature (say) approx 1s BEFORE you start to iron, and set a decrease in temperature back to normal (say) approx 1s before ironing ends.

I have not actually attempted to insert this into Gcode to see what the impact would be, but that would be a way of testing it.

That said, IME the quality of ironing is impacted far more by e.g. ironing two separate areas which adjoin - you tend to get a seam or bump or other imperfection on this.

Despite having raised #7410, I would say that this is probably not worth the effort compared to other things.

Bzzz, 2024-01-14:
Since this is still open, I'd like to +1.
Ironing at 10°C lower than the usual print temperature works well for me; if I run for longer periods of time at full temperature and low extrusion rates like one does for ironing, the nozzle clogs due to heat creep. And while that is generally a hardware issue that could be solved by hardware changes, it's also preventable by software: Lower ironing temperatures, or a higher fan speed setting during ironing, or a quick purge every x minutes / y meters of ironing or other low-throughput activities:
Cura already got several temperature steps for initial layer, final layer, regular layers, small layers and whatnot.
Cura already got several fan settings for initial layer, specific height, bridges and support.
Cura already got a retraction count limiter that watches material flow vs. number of retractions, so it could also monitor material flow vs. time.
Well, I don't think it's unreasonable to ask for another adjustment knob when we already got a thousand ones in Cura 5.6 :)