Ultimaker/Cura #4060: Right sidebar have not enough vertical space to display/view settings

rumatoest, 2018-07-11:
**Application Version**
3.4.0

**Platform**
Linux

**Printer**
Other

**Steps to Reproduce**
Install debian linux with KDE, laucn Cura fullscreen on FullHD resolution (1920x1080)

**Actual Results**
Very tiny area (marked as green) in right sidebar where you can see/edit print settings

**Expected results**
I would expect to have at least 75% of right sidebar height available for settings editing

My point is that "green" area is the main workspace in Cura, but there is almost no space for it.
Look at the area marked with red. It is useless in most cases and you can just hide all of this like "Compatibility mode" features, and show/expand it only by clicking.

[image]

fieldOfView, 2018-07-11:
Have you tried the "Compact Prepare Sidebar" plugin? It is available from the Toolbox.
(NB: can't view your image at the moment, github seems to be having an issue)

rumatoest, 2018-07-11:
Unfortunately I can not see this plugin in already installed.
But I can also see "Could not connect to Cura Package database..." message.

Just downloaded appImage with Cura 3.4.1

BTW. I think this feature have to be packed by default with your application. I could not even guess that some of your plugins can modify existing UI, I thought plugins are for adding extra functionality.

ChrisTerBeke, 2018-07-11:
This has been indicated several times before, and we are working on some re-designs. Note that the compact sidebar plugin modifies the QML structure, so it quite an experimental thing (but awesome nonetheless). We don't ship such plugins by default, because the current design fits in Ultimaker's strategy.

closed 2018-07-11 by ChrisTerBeke

fieldOfView, 2018-07-12:
@rumatoest because the toolbox is not working for you, here is a direct download to the package:
https://api.ultimaker.com/cura-packages/v1/cura/v4/packages/PrepareStageCompactor/download

Download the file and drop it into the Cura window. Then restart Cura and you should have a much more compact sidebar.