Ultimaker/Cura #4061: Could not connect to the Cura package database

rumatoest, 2018-07-11:
**Application Version**
3.3.0 3.4.0 3.4.1

**Platform**
Linux

**Printer**
Other

**Steps to Reproduce**
Toolbox -> Browse packages

**Actual Results**
"Could not connect to the Cura package database"

**Expected results**
Expected to have connection and packages list

diegopradogesto, 2018-07-12:
Check if your network connection is working properly. Check also your firewall is not blocking requests.

rumatoest, 2018-07-12:
Everything works. BTW I can download plugins using direct links from here https://api.ultimaker.com/cura-packages/v1/cura/v4/packages

Maybe something wrong with .AppImage running on Debian.

How can I get logs from your AppImage bundle ?

diegopradogesto, 2018-07-12:
If you go to _Help > Show configuration folder_ and there you can find a **cura.log** file. Please attach it here and maybe we can find something there.

rumatoest, 2018-07-12:
I do not have any cura.log files only ```.  ..  cura.cfg  plugins.json  scripts``` in `.config/cura/3.4/`

rumatoest, 2018-07-12:
Ok I've found logs, but at different location
```
018-07-12 14:50:52,554 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-07-12 14:50:55,639 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 3.1s
2018-07-12 14:51:02,475 - DEBUG - [MainThread] cura.AutoSave._onTimeout [49]: Autosaving preferences, instances and profiles
2018-07-12 14:51:06,047 - INFO - [MainThread] Toolbox.src.Toolbox._makeRequestByType [430]: Toolbox: Requesting packages metadata from server.
2018-07-12 14:51:06,065 - INFO - [MainThread] Toolbox.src.Toolbox._makeRequestByType [430]: Toolbox: Requesting authors metadata from server.
2018-07-12 14:51:06,072 - INFO - [MainThread] Toolbox.src.Toolbox._makeRequestByType [430]: Toolbox: Requesting plugins_showcase metadata from server.
2018-07-12 14:51:06,078 - INFO - [MainThread] Toolbox.src.Toolbox._makeRequestByType [430]: Toolbox: Requesting materials_showcase metadata from server.
2018-07-12 14:51:06,084 - DEBUG - [MainThread] Toolbox.src.Toolbox._createDialog [236]: Toolbox: Creating dialog [Toolbox.qml].
2018-07-12 14:51:06,261 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [301]: file:///tmp/.mount_DPwQcz/usr/bin/plugins/plugins/Toolbox/resources/qml/ToolboxDetailPage.qml:35: TypeError: Cannot read property 'icon_url' of null
2018-07-12 14:51:06,262 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [301]: file:///tmp/.mount_DPwQcz/usr/bin/plugins/plugins/Toolbox/resources/qml/ToolboxDetailPage.qml:58: TypeError: Cannot read property 'name' of null
2018-07-12 14:51:06,264 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [301]: file:///tmp/.mount_DPwQcz/usr/bin/plugins/plugins/Toolbox/resources/qml/ToolboxDetailPage.qml:109: TypeError: Cannot read property 'version' of null
2018-07-12 14:51:06,265 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [301]: file:///tmp/.mount_DPwQcz/usr/bin/plugins/plugins/Toolbox/resources/qml/ToolboxDetailPage.qml:117: TypeError: Cannot read property 'last_updated' of null
2018-07-12 14:51:06,266 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [301]: file:///tmp/.mount_DPwQcz/usr/bin/plugins/plugins/Toolbox/resources/qml/ToolboxDetailPage.qml:127: TypeError: Cannot read property 'author_email' of null
2018-07-12 14:51:06,297 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [301]: file:///tmp/.mount_DPwQcz/usr/bin/qt/qml/QtQuick/Controls/ScrollView.qml:234:9: QML Binding: Binding loop detected for property "value"
2018-07-12 14:51:06,298 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [301]: file:///tmp/.mount_DPwQcz/usr/bin/plugins/plugins/Toolbox/resources/qml/ToolboxDownloadsPage.qml:16:5: QML Column: Binding loop detected for property "width"
```

thattolleyguy, 2018-07-30:
I have the same issue. 

Cura: 3.4.1
OS: Ubuntu 18.04

Let me know if I can provide more information.

ChrisTerBeke, 2018-07-30:
Seems like an SSL issue. We've had these before, but that was using Python's requests module not Qt network. Ping @LipuFei

LipuFei, 2018-07-31:
Saw this before, probably missing SSL libraries in the AppImage.

For Dev: CURA-5608

fcerbell, 2018-07-31:
Same issue with Cura 3.4.0 and 3.4.1. under Debian Stretch
Cura 3.3.1 works

ChrisTerBeke, 2018-07-31:
Yeah, Cura 3.3 doesn't use a secure HTTPS url yet....

fcerbell, 2018-07-31:
Update : 
Does not work on my Debian Stretch upgraded, but works from another stretch freshly installed.

ChrisTerBeke, 2018-07-31:
Is there any difference in the SSL package that's installed on those machines?

fcerbell, 2018-07-31:
Working Stetch : 
~ $ dpkg -l | grep libssl
ii  libssl1.0.2:amd64                     1.0.2l-2+deb9u3                         amd64        Secure Sockets Layer toolkit - shared libraries
ii  libssl1.1:amd64                       1.1.0f-3+deb9u2                         amd64        Secure Sockets Layer toolkit - shared libraries

Non working stretch : 
~ $ dpkg -l | grep libssl
ii  libssl-dev:amd64                       1.1.0f-3+deb9u2                             amd64        Secure Sockets Layer toolkit - development files
ii  libssl-doc                             1.1.0f-3+deb9u2                             all          Secure Sockets Layer toolkit - development documentation
ii  libssl1.0.0:amd64                      1.0.1t-1+deb8u9                             amd64        Secure Sockets Layer toolkit - shared libraries
ii  libssl1.0.0:i386                       1.0.1t-1+deb8u9                             i386         Secure Sockets Layer toolkit - shared libraries
ii  libssl1.0.2:amd64                      1.0.2l-2+deb9u3                             amd64        Secure Sockets Layer toolkit - shared libraries
ii  libssl1.1:amd64                        1.1.0f-3+deb9u2                             amd64        Secure Sockets Layer toolkit - shared libraries
ii  libssl1.1:i386                         1.1.0f-3+deb9u2                             i386         Secure Sockets Layer toolkit - shared libraries

ChrisTerBeke, 2018-08-01:
Maybe it's the libssl-dev packge that is causing the certs not to be verified?

LipuFei, 2018-08-01:
Found a related issue: https://github.com/wkhtmltopdf/wkhtmltopdf/issues/2938
Looks like certain distros have dropped SSL v2 and v3 support completely in their ssl packages. Our build system is CentOS 6.9 which may have the same issue.

alekseisasin, 2018-08-01:
@rumatoest , @thattolleyguy , @tmbdev , @fcerbell . So far we have a solution for this issue, but I cannot perform a good testing because on my machine I have Ubuntu and Cura works fine. I have uploaded Cura (master version) to the internet for testing. Could you please test this build for us and give feedback, whether it works properly or not. 
Link: 
https://wetransfer.com/downloads/870d6184d383d2467c2fdc5fcb11734920180801144239/b9b6a9b30c0e172051bfd09ade3f794820180801144239/b413d2

The link will be available 7 days.

thattolleyguy, 2018-08-01:
Tested and works. I'm able to browse the plugin repository just fine. Thanks!

rumatoest, 2018-08-01:
It works, 

P.S. but it totally screwed up my configuration. Won't start because OpenGL was not found.
I have to reset configuration, and my settings was not saved.

fcerbell, 2018-08-01:
Same here, now I can browse the plugins, select a plugin (tried with Octo print also), install it. It also screwed my configuration. I dont matter, I prefer to have Octoprint ;)

Thanks a lot

LipuFei, 2018-08-01:
I'll create a 3.4.1 build with this fix for you. The master build is not really stable.

ianpaschal, 2018-08-01:
I want to talk to our product owner about making a 3.4.2 release with this and other bug fixes when he returns from vacation next week but can't guarantee that; might have to just wait until 3.5 if master is not that stable right now.

@LipuFei We prefer to not be releasing unofficial builds because it makes it reaaaaally hard to keep track of what version people are using. If you make a testing version to see if this is fixed, make it a temporary link, so that in November we don't have someone running "Lipu's special 3.4.1 build" in a bug report.

Vitaliy79437, 2018-08-01:
Hi, I have the same problem, only on the Windows 7 platform. Help please.

LipuFei, 2018-08-02:
Hi @Vitaliy79437 , I think on Windows 7 it's a different issue, but probably the same issue as https://github.com/Ultimaker/Cura/issues/4099. Could you send us your log file?

Vitaliy79437, 2018-08-02:
Yes, of course, I'll send it now.

Vitaliy79437, 2018-08-02:
[cura.log](https://github.com/Ultimaker/Cura/files/2252775/cura.log)

LipuFei, 2018-08-02:
@Vitaliy79437 Can you see any items in the Toolbox? In the log file, I see that there are connection timeouts for sending anonymous slice data, but no errors regarding the Toolbox. Could you try a different network as well see if it's caused by your current network setup? Thanks.

```
2018-08-02 10:57:08,710 - ERROR - [Thread-4] UM.Logger.logException [81]: Exception: An exception occurred while trying to send slice information
2018-08-02 10:57:08,718 - ERROR - [Thread-4] UM.Logger.logException [85]: Traceback (most recent call last):
2018-08-02 10:57:08,721 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "urllib\request.py", line 1254, in do_open
2018-08-02 10:57:08,730 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "http\client.py", line 1106, in request
2018-08-02 10:57:08,735 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "http\client.py", line 1151, in _send_request
2018-08-02 10:57:08,744 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "http\client.py", line 1102, in endheaders
2018-08-02 10:57:08,747 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "http\client.py", line 934, in _send_output
2018-08-02 10:57:08,749 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "http\client.py", line 877, in send
2018-08-02 10:57:08,751 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "http\client.py", line 1252, in connect
2018-08-02 10:57:08,754 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "http\client.py", line 849, in connect
2018-08-02 10:57:08,756 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "socket.py", line 711, in create_connection
2018-08-02 10:57:08,758 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "socket.py", line 702, in create_connection
2018-08-02 10:57:08,762 - ERROR - [Thread-4] UM.Logger.logException [85]: socket.timeout: timed out
2018-08-02 10:57:08,764 - ERROR - [Thread-4] UM.Logger.logException [85]:
2018-08-02 10:57:08,766 - ERROR - [Thread-4] UM.Logger.logException [85]: During handling of the above exception, another exception occurred:
2018-08-02 10:57:08,769 - ERROR - [Thread-4] UM.Logger.logException [85]:
2018-08-02 10:57:08,771 - ERROR - [Thread-4] UM.Logger.logException [85]: Traceback (most recent call last):
2018-08-02 10:57:08,773 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "...\plugins\SliceInfoPlugin\SliceInfoJob.py", line 32, in run
2018-08-02 10:57:08,775 - ERROR - [Thread-4] UM.Logger.logException [85]:     f = urllib.request.urlopen(self._url, **kwoptions)
2018-08-02 10:57:08,778 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "urllib\request.py", line 163, in urlopen
2018-08-02 10:57:08,781 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "urllib\request.py", line 466, in open
2018-08-02 10:57:08,783 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "urllib\request.py", line 484, in _open
2018-08-02 10:57:08,785 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "urllib\request.py", line 444, in _call_chain
2018-08-02 10:57:08,787 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "urllib\request.py", line 1297, in https_open
2018-08-02 10:57:08,789 - ERROR - [Thread-4] UM.Logger.logException [85]:   File "urllib\request.py", line 1256, in do_open
2018-08-02 10:57:08,797 - ERROR - [Thread-4] UM.Logger.logException [85]: urllib.error.URLError: <urlopen error timed out>
```

Vitaliy79437, 2018-08-02:
Items in the toolbar I do not see at all, I see only those plug-ins that are already installed. Well I'll try a different network. And how do I know if this error is caused by the current settings of my network?

Vitaliy79437, 2018-08-02:
Can not access the site where you can access this access?

Vitaliy79437, 2018-08-02:
Thank you very much, when connecting to a different network, I saw plug-ins.

closed 2018-08-02 by LipuFei

DavidVentura, 2018-09-02:
I had this same issue today:
```
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_num_locks
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_id_callback
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_locking_callback
qt.network.ssl: QSslSocket: cannot resolve ERR_free_strings
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_cleanup
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_init
qt.network.ssl: QSslSocket: cannot resolve sk_new_null
qt.network.ssl: QSslSocket: cannot resolve sk_push
qt.network.ssl: QSslSocket: cannot resolve sk_free
qt.network.ssl: QSslSocket: cannot resolve sk_num
qt.network.ssl: QSslSocket: cannot resolve sk_pop_free
qt.network.ssl: QSslSocket: cannot resolve sk_value
qt.network.ssl: QSslSocket: cannot resolve SSL_library_init
qt.network.ssl: QSslSocket: cannot resolve SSL_load_error_strings
qt.network.ssl: QSslSocket: cannot resolve SSL_get_ex_new_index
qt.network.ssl: QSslSocket: cannot resolve SSLv2_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv3_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv23_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv2_server_method                                                                                                                                                                                                                                                               
qt.network.ssl: QSslSocket: cannot resolve SSLv3_server_method                                                                                                                                                                                                                                                               
qt.network.ssl: QSslSocket: cannot resolve SSLv23_server_method                                                                                                                                                                                                                                                              
qt.network.ssl: QSslSocket: cannot resolve X509_STORE_CTX_get_chain                                                                                                                                                                                                                                                          
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_noconf                                                                                                                                                                                                                                                 
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_conf                                                                                                                                                                                                                                                   
qt.network.ssl: QSslSocket: cannot resolve SSLeay                                                                                                                                                                                                                                                                            
qt.network.ssl: QSslSocket: cannot resolve SSLeay_version                                                                                                                                                                                                                                                                    
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_num_locks                                                                                                                                                                                                                                                 
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_set_id_callback
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_set_locking_callback
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function sk_num
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
```

as @ChrisTerBeke  said
`Maybe it's the libssl-dev packge that is causing the certs not to be verified?`
I uninstalled libssl-dev and now it works