Ultimaker/Cura #4062: Düsengröße lässt sich nicht abspeichern

ViktorHelwer, 2018-07-11:
**Application Version**
Cura 3.4.1

**Platform**
Windows 10 latest

**Printer**
Anycubic Kossel Plus

**Steps to Reproduce**
Unter Einstellungen > Drucker >DeltaBot > Geräteeinstellungen >Extruder > Düsengröße

Die Düsengröße von 0,5 lässt sich nicht verändern! Wenn diese auf 0,4 eingestellt worden ist und Ultimaker beendet wird, bleibt nach dem neustarten der Software Düsengröße bei 0,5!

Wird dieser BUG irgendwann behoben, wenn nicht - mit welcher Software kann man den Code selber verändern?
[image]

diegopradogesto, 2018-07-12:
English please.

fieldOfView, 2018-07-12:
Duplicate of #3338 and #3981

closed 2018-07-12 by ChrisTerBeke