Ultimaker/Cura #4063: Improve Cura start position choice on layers

GreyArea1966, 2018-07-12:
**Application Version**

All versions

**Platform**

All platforms
**Printer**

All printers

**Steps to Reproduce**

Slice a file with Infill layers at overhangs. “Spring no support” on thingiverse is a good example

**Actual Results**

Cura will occasionally start a layer where there is no material beneath it, when there are other options where there WOULD be material beneath.

**Expected results**

Cura should never start a layer in empty space if there is an alternative that is on material laid in the previous layer.

**Additional Information**

A good test piece is “spring, no support” on thingiverse, as it is very small and the layers are almost identical, but rotate...I believe it is this rotation that “confuses” Cura somehow

smartavionics, 2018-07-12:
Related to / duplicate of https://github.com/Ultimaker/Cura/issues/3875.

diegopradogesto, 2018-07-12:
Yes, it looks like it's duplicated. I'll close this to continue the discussing there.

closed 2018-07-12 by diegopradogesto