Ultimaker/Cura #4064: Improve overhang construction

GreyArea1966, 2018-07-12:
**Application Version**
 all

**Platform**

Windows 10
**Printer**

ANet A8
**Steps to Reproduce**

Slicing at overhangs.

Curas algorithm needs improvement in this area and I have a suggestion to achieve it...but it’s not something 8 can code myself. Slice an model with a reasonable overhang and cura does a poor job o constructing it or at least I would suggest can do better.

**Actual Results**

If I gave you ten planks of wood 1m long by 10cm wide, and told you to fix them to reach a ledge 12m away...accepting that you can’t reach the ledge...how would you join them?

When we ask Cur to make an overhang, it effectively joins the planks side to side...so the most it’s going to reach is 10mm.

I suggest that Cura should have a special ‘overhang identifier’ routine with the following rules.

1. If this layer extends beyond the next layer down, then that counts as an overhang.
2. Overhangs should count as separate areas on the layer, extending a distance back from the edge of the previous layer.
3. A routine to identify the direction of the edge of the previous layer at the overhang will be required.
4. Infill setting for overhangs should be as follows;
    a. Infill should be applied first, at a direction that is 90 degrees to the “edge”.
    b. Infill angle on subsequent overhang levels should be +/- 15 degrees (to maintain strength)
    c. overhabg settings override other settings in identified over hang areas
**Expected results**

Example layer #1 8 x 1cm, layer #2 9 x 1 cm, both running north to south, overhang at south end.

Cura identifies the overhang are as the 1cm area, but also 1cm the opposite side (cantilever).
Cura assesses layer #1 and sees the overhang edge is 90 degrees (West to east).
Cura prints main layer areas as normal...So all layer one, an 7cm of layer two.
For the 2x 1cm area of layer 2, Cura starts at the furthest edge that is over layer one material, an applies filament in lines starting on exiting layer #1 material, and running at 180 degrees (90 degrees to edge.
**Additional Information**

I have found far better results when printing overhangs this way, but it is ntirely a manual process...automation would help

closed 2018-07-12 by GreyArea1966

kornshadow097, 2018-07-18:
that sounds interesting...very complicated but so far overhangs seem to fail for me not so much going from center out but coming from out towards center... if this could fix this or improve upon all overhangs that would be nice. I'd also like a proper preview of what the slicer interprets as its print result (S3D does a good job of this, even if its a horrible outlook).

Ghostkeeper, 2018-08-29:
This is effectively in place now with the experimental "Enable Bridge Settings" setting.