Ultimaker/Cura #4065: [3.4.1] Adaptive Layers Variation and Step Size Don't Persist

fdamstra, 2018-07-12:
**Application Version**
3.4.1

**Platform**
Windows 10 x64

**Printer**
Custom

**Steps to Reproduce**
Select a custom profile from the right hand menu.
Enable "Use Adaptive Layers"
Set "Adaptive layers maximum variation" to 0.12
Set "Adaptive layers variation step size" to 0.04
From the "Profile" dropdown, select "Update Profile with current settings/overrides."
Quit cura
Start it again

**Actual Results**
Adaptive Layers maximum variation is set to 0.1 and step size is set to 0.01

**Expected results**
Settings should be saved as 0.12 and 0.04

**Additional Information**
There may be something more to this. Every now and then the setting seems to stick, but then one or two prints later, it's reverted back to 0.1 and 0.01.

jackha, 2018-07-12:
Hi @fdamstra . Thanks for the report. This sounds really strange and I can't reproduce it on Ubuntu. Can you share your preferences folder with us so we can inspect? See: https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations

fdamstra, 2018-07-12:
I think I found a missing piece to the puzzle, and this bug report is inaccurate now. I believe this is actually a case of no profile changes being saved when Cura is launched via Fusion 360. I'm going close this ticket and do more verification on my side to make sure I can consistently get the same results before I open another one.

closed 2018-07-12 by fdamstra

jackha, 2018-07-16:
Thanks in advance for the verification!