Ultimaker/Cura #4074: master branch setting menus are blank

smartavionics, 2018-07-13:
**Application Version**

master

**Platform**

Linux

**Printer**

Any.

**Steps to Reproduce**

Try to alter setting that has a menu.

**Actual Results**

Menu pops up but all entries in the menu are blank.

**Expected results**

Menu entries should not be blank.

**Additional Information**

Material and profile menus are not blank. I think just the setting menus are blank. Here's an example, the menu for Support Interface Pattern is open but blank.

[image]

LipuFei, 2018-07-16:
Hi @smartavionics , I also noticed this recently. Seems like it's only on Linux because I haven't seen this on Mac and Windows. I'm not sure if it has something to do with Qt 5.10 on Linux or some styling issue.

smartavionics, 2018-07-16:
Hi, well if there's anything I can do to help find/fix this, please let me know.

LipuFei, 2018-07-16:
Hi @smartavionics, I've made a fix for this on master e3a193d6a, could you try it out?

smartavionics, 2018-07-16:
That seems to fix the invisible menus, thanks.

BTW, I have just noticed I am getting a bunch of messages like this:

2018-07-16 09:04:11,942 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [354]: file:///opt/cura/share/cura/resources/qml/Settings/SettingComboBox.qml:111:19: QML ItemDelegate: Binding loop detected for property "baselineOffset"

LipuFei, 2018-07-16:
That requires more digging but I don't think we will fix it now...

PR #4086 (2018-07-16): Fix blank dropdown panel in the comboboxes
Align the dropdown panel in the combobox with the parent and get rid of binding loop messages. Fixes #4074.

I did the same fix for the Extruder selector and the Optional Extruder selector. Those were working but for sake of consistency. 

The height of the popups was adjusted so there is no more shakes when hovering the items.