Ultimaker/Cura #4076: Output filename format

SeventhUnderhand, 2018-07-14:
I am looking for a way to set the naming format based on specific variables that would be set in the settings. Just having the printer followed by the object name isn't very descriptive and renaming file can become tedious when slicing a bunch of objects with different settings. 

ex: {Filament_Type}_{Object_Name}_{Infill_Percentage}.gcode

Variable examples:, Infill Percentage, Layer Height, Filament Type (PETG, PLA, ABS, NYLON), Number of objects on the bed, Print Speed, etc.

LipuFei, 2018-07-16:
Hi @SeventhUnderhand , we have planned to let users customize theirs project names via a template which can be set in the Preferences page. For example, `{machine_name}_{object_name}`. However, the list of variables that will be exposed has not been settled yet, so I'm sure if things like "infill percentage" will be possible to put into the template.

For reference, JIRA ticket: CURA-5479

ChrisTerBeke, 2018-07-18:
Duplicate of https://github.com/Ultimaker/Cura/issues/3753?

closed 2018-07-19 by alekseisasin