Ultimaker/Cura #4077: Raft Bottom Layer Count

travisbavar, 2018-07-14:
**Application Version**
3.4.1

**Platform**
MAC OS

**Printer**
CR-10

**Steps to Reproduce**
Set Raft base layer height

**Actual Results**
Not enough filament comes out to stick to the bed

**Expected results**
Base layer height should be what put into input field

**Additional Information**
There are a few heigh end slicing software out there that allow you to set your base layers count similar to the top or middle layer line count. i want my base layer to be high enough that the middle or top layer once extruded will not touch the build plate. I attempt to do this by adjusting the base layer height but i could not get enough filament out fast enough to get a nice line of filament at that height. So i would like for the lines to be able to build on top of eachother similar to how a typical support structure would build and then after a certain height the middle and top layers could build. This feature could help get really stuck prints off of the build plate but stuck well enough that it will not come off

ianpaschal, 2018-07-16:
When I set the raft base height on a CR-10, it appears to work fine.

[image]

Unfortunately we don't have any CR-10 printers in the office to test the actual results with. Can you share some pictures of how your print is turning out with the problem visible?

ianpaschal, 2018-07-24:
Haven't heard back for more than a week and no bug found, so closing.

closed 2018-07-24 by ianpaschal