Ultimaker/Cura #4079: Head stops in front of Wipe and Prime Tower, creating a undesirable plastic hill

francoisba, 2018-07-15:
Hello everyone,

After I read topics about Wipe & Prime Tower management, I still can't find an answer to my problem.

I must say, english is not my mother tongue and I have no idea how to explain what I discovered in _3D printing english_ terms.

I linked [the french version of my issue](https://community.ultimaker.com/topic/23756-param%C3%A8tre-tour-de-purge-pause-inattendue-de-la-t%C3%AAte-dimpression-wipe-prime-tower/), with video and pictures at the bottom.

Well, here is my problem with dual extrusion Wipe & Prime Tower : The head stops right in front of it, instead of going straight to the tower and purge filament.
While the head is waiting for no apparent reason, filament drops or is extruded.
I guess it's extruded based on quantity which leaves from the head even when the output is really stoppered after few layers by the created hill.

I searched in Wipe & Prime Tower settings, Support settings and Temperature settings, but I cannot find a cause for this weird behaviour. It is really strange and new to me, as in previous Cura versions, it didn't happen.

Again, It could be linked to something different than Wipe & Prime Tour settings, so I really need your expertise to find out how to restore a logical behaviour and avoid undesirable wasted amount of material.

Thank you for your attention and your help
Best regards

**Application Version**
3.4.1

**Platform**
Mac OS X10.13
**Printer**
Big Builder Dual Feed

**Steps to Reproduce**
Create a dual extrusion project with wipe & prime tower

**Actual Results**
Head stops in front of the tower, and starts extruding few millimeters filaments.
The result is a catastrophy : after few layer, it creates a hill in front of the tower, and while the head keeps extruding filament, it is already stoppered by previous filament extruded at the same place, right in front of the tower

**Expected results**
The head should not stop, and go straight to the tower to extrude where it is expected to happen, not in front of it

**Additional Information**
https://community.ultimaker.com/topic/23756-param%C3%A8tre-tour-de-purge-pause-inattendue-de-la-t%C3%AAte-dimpression-wipe-prime-tower/

francoisba, 2018-07-16:
I found a lead, it is linked to temperature.

Actually, when it stops, screen displays « heating.. » then « heating done » and printing restarts. I didn’t pay enough attention.

So, I searched for similar issues, and found this one : https://community.ultimaker.com/topic/18820-dual-extrusion-temperatures-nightmare/

As temperatures are the same, I think it’s a mistake to maintain the pause in the code.

And for users printing at two different temperatures, pausing shouldn’t produce an obstacle for the printer’s head.

So, while using swipe and Prime Tower, why don’t make temperature change during the filament purge on the tower, reducing speed if needed.

mtmello2003, 2018-07-18:
I completely agree with you and I´ve been suffering with this same way.

As I can see, cura do not see multi material printers that has a single noozle (cyclops hotend for exemple or maybe "Y" systems).

As a workaround, I created a script to locate all M109 gcode generated and replace it for ;M109. No more wait for temp changes when the tool changes.

francoisba, 2018-07-18:
Hello @mtmello2003 

It actually has to do with **M109 S**XXX, you're right (where XXX stands for targeted temperature)

What seems to work for me is :
opening gcode in a text editor (TextWrangler for example)
Search and delete the "M109 Sxxx" sequence, so it leaves the "M109 T1 Sxxx" sequence, which didn't cause any issue from now.
Then I use search and replace again to erase blank lines _> I have a question about this one_

Is the semicolon sign supposed to _mute_ the following line next to it ?
Or is there something more specific with it, regarding gcode language ?

My question about deleting blank lines in a gcode file :
Could this cause a gcode to lose its points of reference and the printer to consider printing is done while actually there are few layers missing ?

It is a little bit out of topic, but this happened to me for the first time, with the modified gcode, so it could be safer to _mute_ than delete if you didn't notice this kind of bug.

diegopradogesto, 2018-07-19:
Can you please provide a **project file** (File > Save Project...)?
Is the _Big Builder Dual Feed_ one of the printers included in Cura by default?
If you share your project file will be very handy for us, since it includes the machine definition and other files that we need to exactly reproduce your problem. I'm wondering if something is wrong with some settings, and it's just a matter of tweaking them.

francoisba, 2018-07-22:
Hello @diegopradogesto 

Yes, the printer I use is in the default Cura's printers list

Here is a similar project : I used a public stl file, with same printer settings
because I couldn't share the design (I met the issue with a 3Dhubs order)

Hope this can be useful to you guys, I hope you keep us informed.
If you have any other questions regarding this issue, I hope I will be useful.
Good luck

By the way if have any other recommandation to optimise settings, I am willing to try them
Best regards
[dual_extrusion_project.zip](https://github.com/Ultimaker/Cura/files/2217131/dual_extrusion_project.zip)

BagelOrb, 2018-07-26:
Because ultimaker doesn't have any Y merger printers there is not really a reason for Ultimaker to spend time fixing this issue. You will have to hope some community member picks it up.

francoisba, 2018-07-27:
Hi @BagelOrb, I think I knew that already, but thank you for the rudeness of your message.

BagelOrb, 2018-07-27:
Is there a less rude way to provide you with clarity? Please help me become less rude.

francoisba, 2018-07-29:
I appreciate the way you apologise, sincerely.
But anyways, the point here is to share knowledge, and I don't give a damn about your sarcasm.
I'm not here to ask for or provide any help other than regarding 3D printing.

_We_ (the third parts, the _others_) will wait.
But, again, I already provided the solution : So, thank you for your (useless but apparently necessary for you) clarity.

francoisba, 2018-07-29:
Please, it is becoming ridiculous.

Add labels, it is fine then stop here, do not attempt to make fun on me : I asked YOU not you all to leave the "others" alone.
Again, I already knew that : I knew where I am opening a ticket.

When you will understand your clarifications are useless, maybe we can get back to the main topic.
Please, I repeat, you leave us alone.
Regards

mtmello2003, 2018-07-29:
Hello @francoisba , about your question above, yes: semicolon is used to ignore everything after; ie: 
G28 ; Home (just a comment after a g28 command, that will run)

if I do:
;G28 ; Home
Nothing will happen

I think you can delete or comment (";") the line with no problem. Maybe a problem (misinterpretation) can be if you let a blank lines in the file. Better remove them.

diegopradogesto, 2018-07-29:
I was taking a look at this. The behavior is correct from Cura side, since what happen is the following:

- Print with extruder T0 (PVA)
- Go to the **switch position**
- Switch to extruder T1 (PLA)
- Wait until the temperature is reached
- Continue with the print

The problem is that the **switch position** in your printer definition is set to the **prime tower position**. That's why the head stops on top of the prime tower. Probably as the nozzle starts to heat up, some filament falls down (maybe temperature is too high or the PID is not well regulated) and that's why a blob is generated in that position.

I suggest you to change the switch position, by editing the settings `machine_extruder_start_pos_x`, `machine_extruder_start_pos_y`, `machine_extruder_end_pos_x` and/or `machine_extruder_end_pos_y` in the extruder definition files. I never used these printers, so I don't know if they have a specific position for priming, changing tool, ...

These files were added in October 2017 by @Builder3D (https://github.com/Ultimaker/Cura/commit/6dc3b2904797b46f28536c34bbd8c4259a436566), so maybe they want to review these setting to optimize them.

ChrisTerBeke, 2018-07-29:
So it's actually not an issue, but just settings? Closing then.

closed 2018-07-29 by ChrisTerBeke

BagelOrb, 2018-07-29:
I removed the rudeness from the thread.