Ultimaker/Cura #4080: 'NoneType' object has no attribute 'findChild'

jianqingHuang, 2018-07-15:
**Application Version**

**Platform**

ubuntu16.06 x86
**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

Traceback (most recent call last):
  File "/root/uranui/UM/Qt/Bindings/MainWindow.py", line 78, in __init__
    Application.getInstance().setMainWindow(self)
  File "/root/uranui/UM/Qt/QtApplication.py", line 384, in setMainWindow
    self.mainWindowChanged.emit()
  File "/root/uranui/UM/Signal.py", line 212, in emit
    self.__performEmit(*args, **kwargs)
  File "/root/uranui/UM/Signal.py", line 315, in __performEmit
    func(dest, *args, **kwargs)
  File "/root/curaui/cura/../plugins/PostProcessingPlugin/PostProcessingPlugin.py", line 266, in _createView
    Application.getInstance().addAdditionalComponent("saveButton", self._view.findChild(QObject, "postProcessingSaveAreaButton"))
AttributeError: 'NoneType' object has no attribute 'findChild'
**Additional Information**

LipuFei, 2018-07-16:
Hi @jianqingHuang , could you provide reproduce steps for this? Thank you.

jianqingHuang, 2018-07-16:
Hello @LipuFei :
I compiled according to the steps of https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Ubuntu
and this is the version information of the dependent library
cura version:master
Platform:linux-4.13.0-36-gneric-i686-with-ubuntu-16.04-xenial
Qt version:5.5.1
PyQt version:5.8.2
OpenGL:not yes initialized
python3:3.5.2
sip:4.19.2
libprotoc:3.5.1
QMake version 3.0
cmake version 3.5.1

Error traceback

Traceback (most recent call last):
  File "/root/uranui/UM/Qt/Bindings/MainWindow.py", line 78, in __init__
    Application.getInstance().setMainWindow(self)
  File "/root/uranui/UM/Qt/QtApplication.py", line 384, in setMainWindow
    self.mainWindowChanged.emit()
  File "/root/uranui/UM/Signal.py", line 212, in emit
    self.__performEmit(*args, **kwargs)
  File "/root/uranui/UM/Signal.py", line 315, in __performEmit
    func(dest, *args, **kwargs)
  File "/root/curaui/cura/../plugins/PostProcessingPlugin/PostProcessingPlugin.py", line 266, in _createView
    Application.getInstance().addAdditionalComponent("saveButton", self._view.findChild(QObject, "postProcessingSaveAreaButton"))
AttributeError: 'NoneType' object has no attribute 'findChild'

Logs

Thread 0xa4185b40 (most recent call first):
  File "/root/uranui/UM/Backend/Backend.py", line 160 in _storeStderrToLogThread
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0xa4986b40 (most recent call first):
  File "/root/uranui/UM/Backend/Backend.py", line 152 in _storeOutputToLogThread
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0xa5ac3b40 (most recent call first):
  File "/root/curaui/cura/../plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0xacce2b40 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 293 in wait
  File "/usr/lib/python3.5/threading.py", line 424 in acquire
  File "/root/uranui/UM/JobQueue.py", line 86 in _nextJob
  File "/root/uranui/UM/JobQueue.py", line 112 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0xb7c5a980 (most recent call first):
  File "/root/curaui/cura/CrashHandler.py", line 310 in _logInfoWidget
  File "/root/curaui/cura/CrashHandler.py", line 160 in _createDialog
  File "/root/curaui/cura/CrashHandler.py", line 81 in __init__
  File "cura_app.py", line 104 in exceptHook
  File "/root/uranui/UM/Qt/QtApplication.py", line 260 in initializeEngine
  File "/root/curaui/cura/CuraApplication.py", line 796 in runWithGUI
  File "/root/curaui/cura/CuraApplication.py", line 720 in run
  File "cura_app.py", line 137 in <module>

LipuFei, 2018-07-16:
@jianqingHuang We have switched to PyQt 5.10 with Qt 5.10.1 for our master branch to get a few bug fixes. Your PyQt 5.8.2 probably won't work in this case. Since your Qt version 5.5.1 is also outdated, I would suggest to install PyQt 5.10 directly via `pip`, for example, with command `pip install PyQt5==5.10` and try again.

LipuFei, 2018-07-16:
I have also updated the wiki page.

ChrisTerBeke, 2018-07-18:
I had the same issue and after installing qt 5.10 it worked fine again.

jianqingHuang, 2018-07-19:
think you very much @LipuFei @ChrisTerBeke 👍 
This error has disappeared，and it worked fine 👍 
but I have a question ,as the picture shows, the material is empty ,When I modified display name, the interface crashed.
log 👍 
2018-07-19 09:15:01,470 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: An uncaught error has occurred!
2018-07-19 09:15:01,470 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: Traceback (most recent call last):
2018-07-19 09:15:01,470 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/root/Cura/cura/Machines/MaterialManager.py", line 454, in setMaterialName
2018-07-19 09:15:01,471 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     root_material_id = material_node.getMetaDataEntry("base_file")
2018-07-19 09:15:01,471 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'

picture :
[image]

closed 2018-07-31 by jianqingHuang