Ultimaker/Cura #4083: Just upgraded from 3.3.1 to 3.4.1 and no longer slicing objects

Scotty-G5, 2018-07-15:
Just upgraded Cura from 3.3.1 to 3.4.1 on Windows 10 and it is no longer slicing objects. Updated Plugins too. Even tried slicing object that I sliced earlier and printed using 3.3.1.
Tried slicing object using default settings too, still no luck.
Any ideas why Cura 3.4.1 is not slicing?
Thank you

LipuFei, 2018-07-16:
Hi @Scotty-G5 , could you provide the reproduce steps for this? and perhaps a screenshot of the not slicing scenario? I'm not sure what you mean by "not slicing": is it that there is no "prepare" button, or it slices at the beginning but never finishes, or something else?
Thanks!

Scotty-G5, 2018-07-16:
Loaded a simple STL file from thingiverse that I have previously successfully printed, and set up the same profile settings. select the Prepare button and saw a process bar proceed about 1/4 the way and then very quickly disappear. After waiting 5 min, the stl was still not sliced. The Prepare button was changed to a Cancel button.
When loading the same settings and model in 3.3.1, it takes only a second to slice the object and the Prepare button changed to Save to File button.

LipuFei, 2018-07-16:
Hi @Scotty-G5 , looks like CuraEngine crashed. Could you save your failing machine+settings as a project file and share it with us? You can do that via menu "File -> Save as project". Thanks.

Scotty-G5, 2018-07-16:
When saving project it added .3mf extension.  This file type could not be uploaded so I added a .txt to end of file name.

[Yet_Another_Fidget_Infinity_Cube_v2_test.curaproject.3mf.txt](https://github.com/Ultimaker/Cura/files/2197762/Yet_Another_Fidget_Infinity_Cube_v2_test.curaproject.3mf.txt)

LipuFei, 2018-07-16:
Hi @Scotty-G5, looks like you have a Lulzbot TAZ 5 which is not supported in the mainline Cura. On LulzBot's website, they provide a Cura Lulzbot Edition v3.2.21, which I guess is based on Cura 3.2. Because there are many non-compatible changes in Cura 3.3 and 3.4, your lulzbot TAZ 5 machine definition file is not valid any more, and that probably caused the problem.

Thanks for reporting this, but I think you need to contact Lulzbot to update the related files so it can be used in the latest Cura, but at the mean time, I think using Cura Lulzbot Edition v3.2.21 should work for you.

Quacka72, 2018-07-17:
I have the same issue is a pain

LipuFei, 2018-07-17:
@Scotty-G5 @Quacka72 I will close this ticket.

closed 2018-07-17 by LipuFei