Ultimaker/Cura #4087: missing plugins and resources

tecan, 2018-07-16:
latest pull from the repository has missing plugins and resources.

resources/shaders is missing a bunch

plugins/ is missing a bunch like consoleLogger it works ok if i put the plugins from 3.4.1 appimage into a plugins/plugins folder

tecan, 2018-07-16:
nm i found them in uranium

closed 2018-07-16 by tecan

fieldOfView, 2018-07-16:
Noo, dont’t copy (or link) parts of Uranium into Cura! Instead, set the PYTHONPATH env to include Uranium/UM

lf-, 2018-08-31:
I would love not to: issue is I can't seem to figure out how to get it to use Arch's global installations of Uranium in /usr/lib/uranium, specifically the plugins. None of the variations of `PYTHONPATH=/usr/lib/uranium{,/plugins}` work correctly without the `Could not find plugin ConsoleLogger` error.

Ghostkeeper, 2018-10-26:
It can't find a folder called `ConsoleLogger` with a file called `__init__.py` in any of your plug-in locations. Cura's plug-in locations are the following:

* Installation prefix + `lib` + lib suffix + `cura`, for example `/usr/lib64/cura` or probably for you `/usr/lib/cura`.
* Path of CuraApplication.py + `..` + `plugins`, for example `~/projects/Cura/cura/../plugins`.
* Path of UM/Application.py + `..` + `plugins`, for example `~/projects/Uranium/UM/../plugins`.
* Installation prefix + `lib` + `uranium`, for example `/usr/lib/uranium`.
* Installation prefix + `lib32` + `uranium`, for example `/usr/lib32/uranium`.
* Installation prefix + `lib64` + `uranium`, for example `/usr/lib64/uranium`.
* The directory of the current Python executable + `plugins`, for example `/usr/bin/plugins`. This is more of a reason for actual installations that come with a pre-compiled CPython version.
* Installation prefix + `Resources` + `uranium` + `plugins`, for example `/usr/Resources/uranium/plugins`.
* Installation prefix + `Resources` + application name + `plugins`, in Cura's case that is for example `/usr/Resources/cura/plugins`.
* Configuration directory + `plugins`, for example `~/.local/share/cura/plugins`.

Hopefully that should give you enough information on how to debug the problem in your development environment.