Ultimaker/Cura #4091: Cura 3.4.1 Infill Density Bug

kornshadow097, 2018-07-17:
**Application Version**
 3.4.1

**Platform**
Windows 10

**Printer**
 Tevo Tarantula

**Steps to Reproduce**
 Setting infill to zero makes the sliver unable to slice and asks for a "infill line distance" which i have checked as OFF in the settings tab.. however it displays and forces me to input the same value for line width even though i have no infill and its supposed to calculate this on its own... 

**Actual Results**

**Expected results**

**Additional Information**

I have infill set to zero and this shows up. I'd like to see nozzle sizes become part of the print setup menu vs having them stuck on the machine, because i switch nozzles on my machine all the time and dont think it should be considered a different machine due to that.
I'd also like to see a better layer print preview that is animated as layers and as lines. I'd also like better one handed controls on the mouse...having to hold shift is not ideal. I'd like controls similar to that of S3D as well as an option to apply fixes to overhangs and areas that may not work in a print.

I'd also like more options to integrate this with Linear Advance and help determining things like acceleration and jerk settings etc in a simulation environment.

LipuFei, 2018-07-17:
Hi @kornshadow097 , I tried to slice a model with infill density set to 0 and it worked. Could you save your print as a project file and send it to us so we can check what went wrong? Thanks!

You can save as a project file via menu "File -> Save project".

kornshadow097, 2018-07-17:
here you go.

kornshadow097, 2018-07-18:
So does anyone know about this weird issue?

LipuFei, 2018-07-18:
Hi @kornshadow097 , also took us a while to figure it out... The problem is with the "Infill Layer Thickness" but you can ONLY see it if your infill density is larger than 0. So, you can change your infill density to 1 for example, and then you can see this:

[image]

I see that you have "Infill Layer Thickness" set to 0.5 in your custom quality profile. That's why it's using 0.5. The maximum value for it is 0.48 in your case: `layer_height * 8 = 0.06 * 8 = 0.48`.

Although the error message is more or less pointing to the right direction, but it's not straightforward how to correct the values when an error happens. We will discuss see how we can improve that.

Thanks for reporting this!

kornshadow097, 2018-07-18:
I see. Well I don't see why that option should make the slicer fail, especially when the input shouldn't be included in the slice anyway. That's like getting an error for having too thin of a bottom layers when you have bottom layers set to 0.

I see, so layer height *8 is my max infill layer thickness. So then layer thickness isn't the same thing as layer height? I know with a 0.4mm nozzle we arents supposed to go beyond 75-80% or 0.32mm. 

No problem, I enjoy troubleshooting and finding bugs and making suggestion for improvements. Thanks for the help.

I also posted a few suggestions for the app in another post.

Thanks

Jonathan

LipuFei, 2018-07-18:
For infill and support infill, CuraEngine will try to combine multiple layers into one and print that once with a thicker layer height. For example, if you have layer height 0.1mm and infill layer thickness 0.3mm, then 3 layers of 0.1mm infill will be printed on the 3rd layer as a 0.3mm infill layer to save time. This is why the explanation of the setting says that it should be a multiple of the layer height.

The maximum limit for infill layer thickness is there due to a limitation of CuraEngine at the moment: it cannot process too many layers for combining. If you give a higher number, the engine will fail.

kornshadow097, 2018-07-18:
I see. so it actually leaves a infill gap that it will fill up with the combined value later? interesting. 

Where can i find more basics about how cura works? ive been using both cura and s3d and so far s3d i only like due to its ease of use and simplicity but its skin quality and engine is sometimes lacking. while cura has so many options and settings i get overwhelmed with them and sometimes forget the subtle differences in layer types and sizes and max and mins i can and cant use etc...

thanks

Jonathan

LipuFei, 2018-07-18:
I believe @Ghostkeeper knows more than I do. :) You are welcome. I'll close this ticket.

LipuFei, 2018-07-18:
You can also find a lot of discussion in the CuraEngine repository: https://github.com/Ultimaker/CuraEngine

closed 2018-07-18 by LipuFei

Ghostkeeper, 2018-08-29:
> Where can i find more basics about how cura works?

We've been improving the documentation of CuraEngine recently here: https://github.com/Ultimaker/CuraEngine/blob/master/docs/pipeline.md

There has also been an effort to document each setting better, here: https://github.com/alekseisasin/CuraSettingsGuidePlugin

I've fixed the error value for when you've disabled the infill here: 1d8f19f1d1c501f5d7525ca695967a7813f30c16