Ultimaker/Cura #4092: Issue: saving files

ljohnsonnsb, 2018-07-17:
**Application Version**

**Platform**
Windows 7 64bit

**Printer**
all

**Steps to Reproduce**
Rename a project via the interface  and then saving to SD or PC 

**Actual Results**
The renamed project is saved as the original name created by Cura..

**Expected results**
Should be saved as the name I provided.

**Additional Information**

LipuFei, 2018-07-18:
Hi @ljohnsonnsb , which Cura version are you using?

diegopradogesto, 2018-07-18:
Probably a duplicate of #4007 . In this case, it is fixed now on master and it will be available in the next release.

ianpaschal, 2018-07-18:
^I think that's safe to say. Sounds exactly the same. Going to close as fixed so we don't forget about it in September.

closed 2018-07-18 by ianpaschal