Ultimaker/Cura #4093: Wrong Supports.

Dj-EKI, 2018-07-17:
**Application Version**
3.4.1

**Platform**
Windows 10

**Printer**
Anet A8

**Steps to Reproduce**
With 0,8 Nozzle.
[image]

**Actual Results**
Many lines are not supported.

**Expected results**

**Additional Information**
These lines are going in a wrong direction. Layer 29 would have to be across.
[image]

These lines that make a turn are not supported.
[image]

diegopradogesto, 2018-07-18:
We would need some model to see this. Can you please attach the **Project File**. Be sure you use **File > Save Project...** menu.

ianpaschal, 2018-07-18:
It could be possible to try and ensure that the bridging lines run perpendicular to support but for images 2 and 3 there's nothing that can be done. You'd have to make the support run at different intersecting angles everywhere which is not feasible.

Ghostkeeper, 2018-07-19:
The first problem could be solved in theory but we are thinking that it's a lot of work for little gain. An easier solution for the developers is to allow modifying the direction of support lines like we allow this for the direction of infill and skin lines.

M0Dark1, 2018-07-19:
CURA loves to start lines in thin air too. An ability to manually select a layer start point would help to mitigate this, however for what you are showing it wouldn't help. 
Being able to rotate the orientation of an individual support block could be the answer. I assume this is a complex thing for the dev team to achieve.

Dj-EKI, 2018-07-20:
Unfortunately, I can not upload a file because Cura has already reported an error on startup. The profile could not be loaded and was reset. I'll wait until a few more versions have been released and will take another program as long.

Ghostkeeper, 2018-07-25:
You can manually select a layer start point, sort of. It would be the Z seam position. You can set that to Custom.

We decided that we will not be implementing an automated way to rotate parts of your support differently. The complexity of selecting regions to rotate is just too great and you won't get the support lines to align nicely. So that makes this issue a won't-fix.

This issue is still open though as an alternative: https://github.com/Ultimaker/Cura/issues/3481

closed 2018-07-25 by Ghostkeeper