Ultimaker/Cura #4098: Aborted USB print causes loss of connection

M0Dark1, 2018-07-19:
**Application Version**
3.4.1

**Platform**
Windows 10

**Printer**
Creality CR10S

**Steps to Reproduce**
Start a USB print, abort. Attempt another print. 

**Actual Results**
The Cura engine will show the bed and extruder temps coming up while the progress bar moves, when the temps are correct everything will just stop and will sit there indefinitely.

**Expected results**
Temps should come up to correct before progress bar moves, then print should commence.

**Additional Information**
N/A

BagelOrb, 2018-07-26:
This is of no concern to Ultimaker. Let's hope some community member will pick up this bug.

GregValiant, 2024-12-15:
I don't think there is a fix for this.  Aborting prints from a print server is very much firmware and firmware version specific.
I'm cleaning house here and I think this is one that can be closed.  USB Printing has not been updated in a long time and no UM printers use it.  Looking from the outside - trying to deal with all the different USB drivers and hardware, different firmware and printer USB port hardware is a nightmare.
I'll go ahead and close this.

closed 2024-12-15 by GregValiant