Ultimaker/Cura #4099: [3.4.1] Restore backup does nothing

mtmello2003, 2018-07-19:
After trying to restore a backup from one computer to another, cura a message below but never restart by itself and nothing is restored after manual start.

Win7 x64

[image]

mtmello2003, 2018-07-19:
Ok, Cura quit after restore procedure but when I start cura again, nothing were restored. 

Still all old configs.

How to get the log file?

ChrisTerBeke, 2018-07-20:
https://github.com/Ultimaker/Cura#logging-issues

mtmello2003, 2018-07-20:
The sequence is: After start restore procedure in a very short time, cura quit and nothing is restored.

[cura.log](https://github.com/Ultimaker/Cura/files/2213517/cura.log)

ChrisTerBeke, 2018-07-20:
The issue is `(Caused by SSLError(SSLError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:645)'),))`. Your computer seems to be missing then needed SSL certificates.

@LipuFei there are part of the build system right?

LipuFei, 2018-07-24:
@ChrisTerBeke The build system should contain them all. @mtmello2003 Could you try in a browser with this URL https://api.ultimaker.com/cura-drive/v1 and see if you get a certification verification problem?

mtmello2003, 2018-07-24:
@LipuFei I could access the URL w/o problem.

Downloaded a v1 file:
{"uptime": 4925022.269026, "time": "2018-07-24 13:44:00", "version": "0.0.1"}

LipuFei, 2018-07-24:
@ChrisTerBeke I found those two links:
 - https://stackoverflow.com/questions/34646942/ssl3-get-server-certificate-certificate-verify-failed-on-python-when-requesting?rq=1
 - https://github.com/certifi/python-certifi/issues/26
We are using 2018.04.16. I don't know if downgrading will actually work in this case.

LipuFei, 2018-07-24:
Hi @mtmello2003 , discussed with @ChrisTerBeke and we don't really know what caused the problem so far. It could be caused by how the network you are using is set up (for example, some companies may block some domains in their internal network). Could you perhaps switch to a different network and try the plugin again? Thanks.

mtmello2003, 2018-07-24:
Hmmmm... you´re right. Switching to external networking (using 4g modem) I could restore the backup.

Maybe, a FR could solve this kind of issue: option to download the restore file from plugin and restoring using the file downloaded.

ChrisTerBeke, 2018-07-24:
@mtmello2003 although that would solve your situation, the underlying issue is that your router or other part of the network infrastructure is blocking certain SSL certificate checks, even though these certificates are actually valid (from Google in this case).

The plugin and implementation in Cura are both open source, so anyone could add this functionality if the core team doesn't prioritise it.

ChrisTerBeke, 2018-07-24:
Closing because it's not an issue with Cura or the plugin.

closed 2018-07-24 by ChrisTerBeke