Ultimaker/Cura #4100: [3.4.1] Can´t save project

mtmello2003, 2018-07-19:
Win10 x64

After I open a stl file, adjust the configs and try to save the project, get the error:

[image]

fieldOfView, 2018-07-20:
Are you running from source? This probably means that libSavitar is missing.

diegopradogesto, 2018-07-20:
There is no version upgrade to do between 3.4.0 and 3.4.1.

It seems that the `FileWriter` plugins were not properly loaded. Can we get the logs?

mtmello2003, 2018-07-20:
With your answer, everything is clear now: I disabled the 3MF plugins (read and write). Didn´t know the negative impact of that!

Maybe a warning at the plugin description could be usefull.

Right now I enabled 3MF plugins again and the issue vanished.

Tks

diegopradogesto, 2018-07-20:
Oh, great! BTW, I get log entries when a plugin is disabled, I don't know why you don't get anything:

`2018-07-20 13:18:01,814 - DEBUG - [MainThread] UM.PluginRegistry.loadPlugin [382]: Plugin 3MFReader was disabled`

closed 2018-07-20 by diegopradogesto

mtmello2003, 2018-07-20:
It happened ´cause I sent firstly, the log from my work computer and the issue happen with my home computer.

After I sent the log, I see that is about wrong computer and deleted the post but you was faster than that! lol.

Thank you anyway and sorry about that.

mtmello2003, 2018-07-20:
For sure there is no issue in cura due this subject but the only improvement point IMHO is: almost all user do not know that the 3MF files is used by cura to save projects.

In my needs, I use only STL or OBJ files; never 3MF... thats the way I disable those plugins...

diegopradogesto, 2018-07-20:
That's why we call our project files _*.curaproject.3mf_

:-P

fieldOfView, 2018-08-31:
Even if the plugin does not load, the File -> Save menu is still there. The plugin not loading does not solve the problem.