Ultimaker/Cura #4110: superfluous support

Timechild2013, 2018-07-22:
Hello, I have always had this problem with redundant supports.
It is always generated on the side where it is not needed. No matter what settings I make, nothing changes. This problem has been around for a very long time.
Printer is Prusa i3, Windows 10 pro 64bit, Cura 3.4.1

`https://drive.google.com/open?id=1Ww7CzmBNPBaRDYF5KWdnQMOX1ya2GzE5`
`https://drive.google.com/open?id=1_gyqC5nrXSelfQKZ4leXKWWLrhBm1tCw`

diegopradogesto, 2018-07-23:
Thanks for providing pictures and project file. We will analyse it.

diegopradogesto, 2018-07-24:
You can get rid of those undesirable supports by turning off the _Use Towers_ setting or reducing the _Tower Diameter_ if the former setting needs to be enabled.

Timechild2013, 2018-07-24:
Wow, that was still classified as an "experiment" and at that time I had tried without knowing the impact and since then had been so relaxed and had this problem again and again. All these years a "scrap" printed. Would not it be better, everything that is hidden is automatically disabled? Who should know and understand all of their functions? after all, there is no manual. Although there are hints for the function, but who should know what it does?

But thanks for your help!

diegopradogesto, 2018-07-24:
This setting is enable by default because it helps in the majority of the cases. Maybe this specific model doesn't need it, but we can't control model dependent settings (yet).

closed 2018-07-24 by diegopradogesto