Ultimaker/Cura #4111: [3.4] Windows Installer, Installs 64bit version in 32bit registry

damon-atkins, 2018-07-22:
**Application Version**
3.4.*

**Platform**
Windows

**Steps to Reproduce**
Install the software e.g.  Ultimaker%20Cura-3.4.1-win64.exe
**Actual Results**
Install Data placed in
Computer\HKEY_LOCAL_MACHINE\SOFTWARE\WOW6432Node\Microsoft\Windows\CurrentVersion\Uninstall\Ultimaker Cura 3.4

**Expected results**
Install Data should be in
Computer\HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\CurrentVersion\Uninstall\Ultimaker Cura 3.4

Are you using a 32bit installer to install a 64bit program?

**Additional Information**
Consider also setting in the registry the following options:
- URL values  `URLInfoAbout`, `HelpLink`, `MoreInfoUrl`
- `InstallLocation` - path to install location
- `Comments` - text
- `QuietUninstallString` - same as UninstallString with /S added

[image]

fieldOfView, 2018-08-01:
The installer is a 32 bit application, the uninstaller is too.

damon-atkins, 2018-08-01:
That will explain why its going into the wrong spot in the registry. Please start using a 64 bit version of your installer for 64bit app.

damon-atkins, 2018-08-01:
I think you need to use 
`!include "x64.nsh"`

damon-atkins, 2018-08-08:
ping @alekseisasin and @awhiemstra

fieldOfView, 2018-08-08:
Note that https://github.com/Ultimaker/Cura/blob/master/installer.nsi has not been used since september 2015 (and could be removed if you ask me). Instead Cura is built using https://github.com/Ultimaker/cura-build (see https://github.com/Ultimaker/cura-build/blob/master/cmake/NSIS.template.in for the NSIS script)

damon-atkins, 2018-08-08:
Seems `!include "x64.nsh"` is missing in that as well.  And it seems to also include a little bit of 32 bit checking i.e.
```
  IntCmp $vcredist_selected 0 +2
  ExecWait '"$INSTDIR\vcredist_x@CPACK_NSIS_PACKAGE_ARCHITECTURE@.exe" /install /quiet /norestart'

  IntCmp $arduino_selected 0 +6
  ${If} ${RunningX64}
    ExecWait '"$INSTDIR\arduino\dpinst-amd64.exe" /lm /sw'
  ${Else}
    ExecWait '"$INSTDIR\arduino\dpinst-x86.exe" /lm /sw'
  ${EndIf}

@CPACK_NSIS_EXTRA_INSTALL_COMMANDS@
```
Which most likely should be removed given only 64bit versions are only being released.

damon-atkins, 2018-09-03:
BuildPlatformDetection.cmake contains
```
elseif(WIN32)
    set(BUILD_OS_WINDOWS ON)
    if(CMAKE_SIZEOF_VOID_P EQUAL 8)
        set(BUILD_OS_WIN64 ON)
    else()
        set(BUILD_OS_WIN32 ON)
    endif()
```

From cura-build/packaging/setup_win32.py.in
```
    if platform.architecture()[0] == "32bit":
        required_mkl_files.extend([
            "mkl_p4.dll",
            "mkl_p4m.dll",
            "mkl_p4m3.dll",
        ])
    else:
        required_mkl_files.extend([
            "mkl_def.dll",
            "mkl_mc.dll",
            "mkl_mc3.dll",
        ])
```
cura-build/packaging/windows.cmake
```
if(BUILD_OS_WIN32)
    install(FILES ${EXTERNALPROJECT_INSTALL_PREFIX}/vcredist_x32.exe
            DESTINATION "."
            COMPONENT "vcredist"
            )
else()
    install(FILES ${EXTERNALPROJECT_INSTALL_PREFIX}/vcredist_x64.exe
            DESTINATION "."
            COMPONENT "vcredist"
            )
endif()
.......
if(BUILD_OS_WIN32)
    set(CPACK_NSIS_PACKAGE_ARCHITECTURE "32")
else()
    set(CPACK_NSIS_PACKAGE_ARCHITECTURE "64")
endif()

set(CPACK_NSIS_PACKAGE_NAME ${CPACK_PACKAGE_NAME})

include(CPack)
```

Seems like this needs some consistency.  i.e. include(CPack) I assume should include all ARCH related statements, yet vcredist is in windows.cmake.

Ghostkeeper, 2018-11-05:
> Note that https://github.com/Ultimaker/Cura/blob/master/installer.nsi has not been used since september 2015 (and could be removed if you ask me). Instead Cura is built using [...]

Removed that. Thanks.