Ultimaker/Cura #4112: Warning : Could not save to Users/Desktop/file.gcode : none

is05458, 2018-07-22:
I'm using cura 3.4.1 at Mac and Windows 7 but I have the same error warning when I try to export a model to gcode. It happens with all models.

If I use Cura 15.04.6 with Windows 7 I don't have this issue and I can export correctly the gcode file. But I want to use the last version with more parameters.

I reviewed the log application but it doesn't seem to help.

**Application Version**
3.4.1

**Platform**
macOS Sierra & Windows 7

**Printer**
Anycubic i3 Mega

**Actual Results**
Warning : Could not save to "file" : none.

diegopradogesto, 2018-07-23:
This error is raised when there is a PermissionError, so probably you are trying to save to a location where you don't have permissions. Try another folder and see if it works.

is05458, 2018-07-23:
Hi Diego,
I've tried on several folders with the same result.
I've uninstalled 3.4.1 and installed 3.3.1 and it works fine. So I don't think that's a folder permission issue.

Thx!

ianpaschal, 2018-07-26:
^That doesn't really prove that it's not. Can you try what Diego suggested anyway?

rxharr6, 2018-08-02:
I had this issue when I tried to save before selecting "prepare" first time I tried to use the app.

BagelOrb, 2018-08-02:
@rxharr6 is right. This error happens when you try to export to gcode before you have sliced.

- Turn off slice automatically
- load model
- File > Export ... > select gcode > selection location where you could save the gcode.

BagelOrb, 2018-08-02:
One choice, two options:

- make export as gcode unselectable when the build plate is empty or when a setting has changes (just like layer view)
- Selecting Export > gcode slices for you wit hthe current settings.

ianpaschal, 2018-08-03:
We discussed this as a team and we're not going to change the behavior here. We also don't really consider it a bug because saving an empty gcode file before you've generated gcode is very much "expected behavior."

We thought it might be nice to also not generate the empty file but it quickly became obvious we'd need to write a pre-writer for the file since we don't know that the file job failed until after creating the file on disk, attempting to put gcode into it, and failing. This is too much work for not enough gain.

In the process we did discover another bug though which is that if you generate gcode, then move your model, the old gcode is not erased from memory until new gcode replaces it, meaning if you forget to hit prepare you might save the old gcode a second time thinking it's taking the new position into account.

This one is an easy fix, so we went ahead and changed it.

ianpaschal, 2018-08-03:
Change was made in #4185

closed 2018-08-03 by ianpaschal

fieldOfView, 2018-08-03:
I am (still) convinced that saving geometry/projects and saving sliced results (gcode or ufp) are two entirely different things and should not be included in the same "Save as..." command. If they were split into two commands, this issue would be a lot simpler to fix.

ianpaschal, 2018-08-03:
This is true, and we've been discussing lately how to break apart the various "save", "save as", "export" nomenclature, but we consider this ticket to be a more distal issue.

BagelOrb, 2018-08-03:
Well it is in some sense expected behavior, but it is still bugging new users. For them it will be a mystery why their gcode doesn't do anything when they give it to the machine. I would still consider this a bug - even if it is a minor bug.

diegopradogesto, 2018-08-03:
We just add this PR https://github.com/Ultimaker/Uranium/pull/374 that shows some information to the users when something went wrong saving files. I know that it doesn't solve the problem, but at least the users will be aware that the gcode is empty.

Ghostkeeper, 2018-09-28:
The splitting of nomenclature is implemented in Cura 3.5. See if you think that this fixes the issue.

BagelOrb, 2018-09-28:
If I now try the steps it gives an error "Please generate gcode before saving."
Good!

Little remark: I think it should be "Please **prepare** gcode before **exporting**."
The button at the bottom right is called [Prepare] and the item selected in the menu is called [Export...]

Ghostkeeper, 2018-09-28:
Agreed. I've corrected that text.

lquochuy, 2019-05-11:
It is simply to click on 'Slice' button on the right bottom then it allows to save.